ros2/rosbag2 #1502: Resort: When playing packets in ROS2 Humble, the bandwidth is very low

zxl152723, 2023-11-23:
Hardware Lenovo Saver y9000P2023 CPU i9-13900hx 16G memory solid-state drive (Western 770 black drive), software environment is ubuntu22.04. Problem: After playing the bag, 
use ROS2 topic bw/rslidar_ point check bandwidth (only a few hundred kb/s), as shown in the following figure.
But on another computer, the normal speed can reach 7-8M/s, may I ask why?
[image]

MichaelOrlov, 2023-11-28:
@zxl152723 Hmm, don't know why off the top of my head.  
If thinking logically, if you see the correct bandwidth receiving on another computer it does mean that rosbag2 player publishing with the correct rate, right? 
It means this is not a rosbag2 bug, correct?

Might be default QoS settings on the receiver side or the maximum allowed buffer in the kernel for the Ethernet adapter.

zxl152723, 2023-11-29:
Thank you for your help，I have another question. When I play the ros2 bag ，the hz of the topic “/rslidar_points” is so low, it should be 10Hz, but in my device it only 0.9hz，how can I set the Qos？

when I playing the bag in my device and using the “”ros2 topic hz /rslidar_points” to view  with another device ，the hz is about 10Hz，why？

MichaelOrlov, 2024-01-30:
- Closing a stale issue without the possibility of reproducing it.

closed 2024-01-30 by MichaelOrlov