ros2/rosbag2 #1508: param set uri doesn't work for Player component

KenYN, 2023-12-01:
## Description
With the new component Player, I cannot change the file to play back

### Expected Behavior
Given this YAML file:
```
launch:
    - node_container:
        pkg: rclcpp_components
        exec: component_container
        name: my_container
        namespace: ''
        composable_node:
            -   pkg: rosbag2_transport
                plugin: rosbag2_transport::Player
                name: bag_player
                namespace: ''
                param:
                    - name: uri
                      value: '/path/to/file1/'
```
Then the following commands:
```
$ ros2 param set /bag_player uri "/path/to/file2"
$ ros2 service call /bag_player/play rosbag2_interfaces/srv/Play "{start_offset: {sec: 0, nanosec: 0}, playback_duration: {sec: 65535, nanosec: 0}, playback_until_timestamp: {sec: 0, nanosec: 0}}"
```
(Note, `Play` doesn't seem to have a `play_to_end` value)

Should result in `file2` being played back.

### Actual Behavior
`file1` is played back, even though:
```
$ ros2 param get /bag_player uri
String value is: /path/to/file2
```

## To Reproduce
See above

## System (please complete the following information)
 - OS: Ubuntu 22.04
 - ROS 2 Distro: rolling
 - Install Method: source
 - Version: patch-2

MichaelOrlov, 2023-12-01:
@roncapat Any ideas why `ros2 param set /bag_player uri "/path/to/file2"` doesn't work as expected?

roncapat, 2023-12-01:
Well, my first thought is that we are not handling any parameter update IMHO. Parameters are just parse one time at the beginning and stop. 

Example:
```cpp
storage_options.uri = node.declare_parameter<std::string>("uri", "");
```

after this line, `storage_options.uri` is not bound to any `rclcpp::Parameter`_something_ or stuff like that. It is a simple `std::string`. And nothing is done currently to support updated on the fly.

I will think a bit about how this could be done.

MichaelOrlov, 2023-12-02:
@KenYN @roncapat I think we will close this issue as unsupported by design.

@roncapat BTW as regards
> (Assuming we can get notified by something that a parameter has changed).

Do you have ideas on how we can potentially get notified when the parameter got changed?

roncapat, 2023-12-02:
There seems to be the possibility to add a callback.
 https://roboticsbackend.com/ros2-rclcpp-parameter-callback/

MichaelOrlov, 2024-05-05:
- Closing this issue as functionality is currently unsupported by design.

closed 2024-05-05 by MichaelOrlov