ros2/rosbag2 #1511: Play folder of ROS2 bags in sequential order or let them be fed as input arguments.

lowellm91, 2023-12-14:
## Description

I have a folder of ROS2 bags which came from a single bag file, they are numbered sequentially. I want to have them play in sequential order. Some people said this worked in ROS1 but I haven't figured out how to make it work in ROS2. In addition I would like the same behavior if they are fed as arguments to ros2 bag in sequential order. 

## Related Issues

None as far as I know. 

## Completion Criteria

If there is a way to get ROS2 to play a sequential set of bags(either from a folder) or from command line.

## Implementation Notes / Suggestions

## Testing Notes / Suggestions

Provide a folder of numbered bag files and play them sequentially.

MichaelOrlov, 2023-12-14:
Hi @lowellm91 Yes rosbag2 supports such functionality as well.
You can manually edit `metadata.yaml` file to specify your files and their order.
See the example in the 
https://github.com/ros2/rosbag2/blob/0fcc839d4363855fe4a5c8693b5e16dfd64b9b34/rosbag2_py/test/resources/mcap/wbag/metadata.yaml#L1-L12

However, there is one caveat. The timestamps inside each bag file shall be in sequence relative to the previous file.

closed 2023-12-14 by MichaelOrlov

lowellm91, 2023-12-14:
I believe my metadata file has this except each bag has a db3 extension, when I ros2 play with the directory which contains the bags and the metadata the first bag file plays then it stops. 
[metadata.txt](https://github.com/ros2/rosbag2/files/13669007/metadata.txt)

MichaelOrlov, 2023-12-14:
@lowellm91 If you add them manually you may need to adjust other fields as well:
```
duration:
    nanoseconds: 1149684188
  starting_time:
    nanoseconds_since_epoch: 1700087500680805153
  message_count: 96056
```
As regards to your observation:
> when I ros2 bag play with the directory which contains the bags and the metadata all the files open at the same time and it appears the only first one plays.

It can't be that all files opened a the same time. From your screenshot it looks like it was played back the first file and the rest files was opened-closed and switched to the next file since the timestamps was in wrong order or total message count or total bag duration was reached the limit settled up in the metadata.