ros2/rosbag2 #1512: All bags being opened when playing back multiple bag files.

lowellm91, 2023-12-14:
I requested a feature to play multiple bags in a single folder sequentially. It was mentioned that this feature is present in rosbag2 which means there is an error going on(I hope it's OK to create a second thread). As can been seen when I play the folder every bag is opened but only the first bag appears to output via echo. I checked using rqt if the time-stamps between each file were increasing and they are so I'm not sure what could be the hangup.. 

              [image]
I believe my metadata file has this except each bag has a db3 extension, when I ros2 bag play with the directory which contains the bags and the metadata all the files open at the same time and it appears the only first one plays. 
[metadata.txt](https://github.com/ros2/rosbag2/files/13669007/metadata.txt)

_Originally posted by @lowellm91 in https://github.com/ros2/rosbag2/issues/1511#issuecomment-1855166929_

MichaelOrlov, 2023-12-14:
@lowellm91 We need a minimum valuable reproducer for this problem.

lowellm91, 2023-12-14:
> @lowellm91 We need a minimum valuable reproducer for this problem.

Thank you, I will check and see if I can provide one. I know the bag file was recorded with Foxy and then split into a collection of smaller bags for easier transmission. I am using Humble - could this be the problem?

MichaelOrlov, 2023-12-14:
>  I am using Humble - could this be the problem?
It shouldn't be a problem we are trying to keep backward compatibility in the new versions of the rosbag2.

lowellm91, 2024-01-24:
Hi, I apologize as I was not able to provide the data, I will close the issue as it might be related to the bag files themselves.

closed 2024-01-24 by lowellm91