ros2/rosbag2 #1519: TimeControllerClock::sleep_until(some_time_in_the_past) sleeps too long

KenYN, 2023-12-21:
## Description
When processing a high-frequency file (eg 5Gbps split into 1400 byte messages, or one message every 2.2 microseconds) `PlayerImpl::play_messages_from_queue()` starts publishing at slower than real-time

### Expected Behavior
I would expect publishing to happen in realtime.

### Actual Behavior
Extensive debugging pinned down the culprit to this function call:

https://github.com/ros2/rosbag2/blob/cea3fb027e52b9445266adde9d7629b9957a3c35/rosbag2_transport/src/rosbag2_transport/player.cpp#L852

If the message header time is behind the current Sim time, rather than returning immediately as there is no waiting to be done, it instead calls this line, which takes 2.2 microseconds to execute:

https://github.com/ros2/rosbag2/blob/cea3fb027e52b9445266adde9d7629b9957a3c35/rosbag2_cpp/src/rosbag2_cpp/clocks/time_controller_clock.cpp#L171

According to [Stack Overflow](https://stackoverflow.com/q/70375285/1270789):

> Thus, wait_until() unconditionally releases the lock, and even if it unblocks right away on account of timeout, it must then reacquire the lock. This leaves open the possibility that another thread acquires the lock in between, in which case that other thread can hold the lock for an arbitrary amount of time, delaying the return of the wait_until().

I wonder if something other thread is grabbing the lock, or is a context switch stalling the thread?

## To Reproduce
Try playing a high-frequency file - I'd attach my test file, but it's over 3GB in size.

## To Fix

Before the line:

https://github.com/ros2/rosbag2/blob/cea3fb027e52b9445266adde9d7629b9957a3c35/rosbag2_cpp/src/rosbag2_cpp/clocks/time_controller_clock.cpp#L171

I added code that skips that call if we have already passed `until`. This hack works for me, but I'm unable due to company policies, etc, to work it into a PR, and I've not shown any code just in case my obvious one-liner causes any licensing issues.

## System (please complete the following information)
 - OS:Ubuntu 22.04
 - ROS 2 Distro: Rolling
 - Install Method: Source
 - Version: With Player composable node patches

MichaelOrlov, 2023-12-23:
@KenYN The first questions which arise reading the description of the issue, - Do you use RTOS of Linux with RTOS patch?
And what is the maximum context switch time in the OS?

2.2 microseconds is a pretty tough loop. And most likely you see playback not in realtime due to the context switches or some other atomic primitives overhead during `cv.wait_untill()`. 
As regards to your question 
> I wonder if something other thread is grabbing the lock, or is a context switch stalling the thread?

In the default configuration, there are no other threads that are competing for the lock in the continuous manner unless the `PlayOptions::clock_publish_frequency` settled up more than 0. We can try to lock the same mutex only when pause/resume called or `clock_->now()` method called.

Introducing check for `if (steady_until > impl_->now_fn())` before  `impl_->cv.wait_until(lock, steady_until);` might help just because `wait_untill` will not be called in most of the cases with 2.2 microseconds loop. But more appropriate would be to use RTOS with the context switch way below 2.2 uSec.

I am also curious about. Ok, let's hypothetically assume that we solved delays in `impl_->cv.wait_until(lock, steady_until);` or overhead delays there don't exist. How to guarantee that publishers will be able to publish messages in time less than 2.2 uSec? 
Inside the publisher likely will be the same problem or even worse.

cc: @fujitatomoya

KenYN, 2023-12-25:
@MichaelOrlov I'm using a realtime kernel, but at the moment I'm not looking for hard realtime playback - my debugging revealed that the reason I was not getting the expected 5 Gbps was due to `wait_until()` taking too long, so I reported it here as a bug as others might encounter the same behaviour.

I've addressed other issues like `GenericPublisher` not being able to support zero copy and DDS bypass by deserialising known topics within the  `Player` component, but that's just a private hack, of course.

MichaelOrlov, 2023-12-30:
- Closing as fixed in the https://github.com/ros2/rosbag2/pull/1521

closed 2023-12-30 by MichaelOrlov

MichaelOrlov, 2024-02-05:
@KenYN Out of curiosity, how did you manage to record such high-frequency file?  
> (eg 5Gbps split into 1400 byte messages, or one message every 2.2 microseconds)

KenYN, 2024-02-06:
@MichaelOrlov I generated them programmatically.