ros2/rosbag2 #1520: Can't play bags with v9 metadata

roncapat, 2023-12-22:
When recording a bag after merge of https://github.com/ros2/rosbag2/pull/1476, and you later want to replay it, `metadata.yaml` parsing fails. Most probably, we do not have in place a test that records and then plays back (?).

See also https://github.com/ros-visualization/rqt_bag/issues/146 since I noticed this by starting to test the `rolling` branch on a real project and initially thought that the problem was in `rqt_bag`.

roncapat, 2023-12-22:
Sorry for the noise... I tested `rosbag2` `rolling` in an `iron` distribution... and apparently something doesn't work as it should be in this mix. Rolling correclty works fine.

closed 2023-12-22 by roncapat

MichaelOrlov, 2023-12-22:
> I tested rosbag2 rolling in an iron distribution... and apparently something doesn't work as it should be in this mix. 

Likely because of ABI incompatibility. We don't guarantee ABI compatibility between ROS 2 distros.