ros2/rosbag2 #1527: Qos question

SunGod1202, 2024-01-04:
New publisher discovered on topic '/demo/laser/out', offering incompatible QoS. No messages will be received from it. Last incompatible policy: RELIABILITY

clalancette, 2024-01-04:
Please open questions like this on https://robotics.stackexchange.com/questions/tagged/ros, which is our central Question and Answer site.  You'll get a better answer there, and it will be searchable for the future.

Make sure to include a lot of information on what platform you are using, which ROS distribution you are using, and the exact steps you took.

closed 2024-01-04 by clalancette

SunGod1202, 2024-01-06:
I have resolved this issue by handling it this way. Thank you for your assistance.

elf.subscription = self.create_subscription(
            Odometry,
            '/odom',
            self.listener_callback,
            rclpy.qos.qos_profile_sensor_data)