ros2/rosbag2 #1528: Capture implicit message definitions for action feedback messages

taughz, 2024-01-05:
## Description

I would like to be able to record a bag using `--include-hidden-topics` and have it capture implicit message definitions for action feedback messages, e.g., `action_tutorials_interfaces/action/Fibonacci_FeedbackMessage`. I know service introspection is on the way for Jazzy, but it's unclear to me if service introspection solves this problem. I'm currently on Humble and using `-s mcap`; I also tried Iron and Rolling, but got the same error message:

```
[ERROR] [1704486429.414617621] [rosbag2_cpp]: Message type name 'action_tutorials_interfaces/action/Fibonacci_FeedbackMessage' not understood by type definition search, definition will be left empty in bag.
```

It looks like work on this feature was started in #1341, but then that effort was stopped. (Was there another issue or PR that replaced that one?)

## Related Issues

* https://github.com/ros2/rosbag2/issues/1286
* https://github.com/ros2/rosbag2/pull/1341

## Completion Criteria

* The implicit message definition is somehow captured and stored in the bag.
* No longer see the `not understood by type definition search` error message.

## Implementation Notes / Suggestions

* None.

## Testing Notes / Suggestions

* None.

ryank-cobot, 2024-02-27:
I would also love this functionality!

MichaelOrlov, 2025-05-07:
The action recording has been done in the https://github.com/ros2/rosbag2/pull/1939 and is available in Kilted ROS 2 distro.
However, storing message definition for `action/Fibonacci_FeedbackMessage` is going to be handled in the 
 https://github.com/ros2/rosbag2/issues/1966