ros2/rosbag2 #1529: `Recording stopped` message is duplicated when stop recording

fujitatomoya, 2024-01-05:
## Description

`Recording stopped` is printed twice when we stop recording with `ros2 bag record...`

### Expected Behavior

it should print only once.

### Actual Behavior

```
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 bag record --all-topics
[INFO] [1704478296.853012766] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1704478296.865478656] [rosbag2_recorder]: Listening for topics...
[INFO] [1704478296.865513875] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1704478296.870645864] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1704478296.872183975] [rosbag2_recorder]: Subscribed to topic '/events/write_split'
[INFO] [1704478296.872236186] [rosbag2_recorder]: Recording...
[INFO] [1704478296.995752165] [rosbag2_recorder]: Subscribed to topic '/chatter'
[INFO] [1704478297.006186393] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
[INFO] [1704478305.969940135] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
[INFO] [1704478305.975847475] [rosbag2_recorder]: Event publisher thread: Exiting
[INFO] [1704478305.975882003] [rosbag2_recorder]: Recording stopped
[INFO] [1704478305.977988895] [rosbag2_recorder]: Recording stopped
```

## To Reproduce

`ros2 bag record ...`

## System (please complete the following information)

 - OS: Ubuntu 22.04
 - ROS 2 Distro: Rolling
 - Install Method: Source build
 - Version: https://github.com/ros2/ros2/commit/ed4995259068d8e8a3c6eb586f4bc9a848dfbc89

fujitatomoya, 2024-01-10:
### Implementation Consideration

I think we can add the state control mutex lock (or rwlock, probably not needed since atomic operation can be enough to read the data such as `RecorderImpl::is_paused()`) aligned with `RecorderImpl::paused_` and `RecorderImpl::in_recording_`. i guess that we need to add some state checks at the beginning of methods.
this should avoid this original problem since if the recorder is already stopped, it does not have to do the stop procedure again.

MichaelOrlov, 2024-01-30:
- Closing a done in the #1530

closed 2024-01-30 by MichaelOrlov