ros2/rosbag2 #1532: How to read a mcap file and publish data using cpp?

yang19900808, 2024-01-09:
Is there any one who knows that how to read an existing mcap file(recorded by ros2) using cpp and then publish messages in ros2?
In the official example, I just find a writer demo and reader demo in other language.

MichaelOrlov, 2024-01-09:
Please open questions like this on https://robotics.stackexchange.com/questions/tagged/ros, which is our central Question and Answer site. You'll get a better answer there, and it will be searchable for the future.

MichaelOrlov, 2024-01-09:
It will be something like that..
```cpp
rosbag2_storage::StorageOptions storage_options;
  storage_options.uri = "path/to/the/bag/file/or/folder";
  storage_options.storage_id = "mcap";
  auto reader = rosbag2_transport::ReaderWriterFactory::make_reader(storage_options);
  reader->open(storage_options);
  auto filter = rosbag2_storage::StorageFilter{};
  std::string topic_name = "topic_name_to_publish";
  filter.topics.emplace_back(topic_name);
  reader->set_filter(filter);

  const auto metadata = reader->get_metadata();

  std::string topic_type;
  rclcpp::QoS topic_qos_profile{1};

  for (const auto & topic : metadata.topics_with_message_count) {
    if (topic.topic_metadata.name == topic_name) {
      topic_type = topic.topic_metadata.type;
      topic_qos_profile = topic.topic_metadata.offered_qos_profiles[0];
    }
  }

  std::shared_ptr<rclcpp::GenericPublisher> pub =
    node_->create_generic_publisher(topic_name, topic_type, topic_qos_profile);

  while (reader->has_next()) {
    auto msg = reader->read_next();
    pub->publish(rclcpp::SerializedMessage((*msg->serialized_data)));
  }
```

closed 2024-01-09 by MichaelOrlov

yang19900808, 2024-01-10:
@MichaelOrlov thank you for the answer.
I have tried your code but I think some interfaces are not existing in foxy.
I post [a new question,](https://robotics.stackexchange.com/questions/107510/how-to-read-a-mcap-file-and-publish-message-using-cpp-and-ros2foxy) could you tell me more about it?

MichaelOrlov, 2024-01-10:
@yang19900808 Thanks for the question on the stack exchange.
Unfortunately, there are no solutions for Foxy because there is no support for MCAP file format in the Foxy distro and it has already reached the EOL.

We added support for MCAP file format starting from Humble.