ros2/rosbag2 #1549: Subscribers to topics don't record data.

D-2n, 2024-01-22:
Hello. I am trying to write a python script in order to record a ros2 bag, however, only data from the last topic in topics list is being recorded. I implemented it this way as all the examples online are single-topic recorders.

Any ideas about why this doesn't work?
`#!/usr/bin/env python3
# -*- coding: utf-8 -*-

import rclpy
import inspect
from rclpy.node import Node
from rclpy.serialization import serialize_message
from std_msgs.msg import String
from sensor_msgs.msg import Image, CameraInfo
import datetime
import rosbag2_py

topics = [
    ['/carla/ego_vehicle/rgb_view/image', 'sensor_msgs/msg/Image', Image],
    ['/carla/ego_vehicle/rgb_front/image', 'sensor_msgs/msg/Image', Image],
    ['/carla/ego_vehicle/rgb_front/camera_info', 'sensor_msgs/msg/CameraInfo', CameraInfo],
    ['/carla/ego_vehicle/rgb_view/camera_info', 'sensor_msgs/msg/CameraInfo', CameraInfo]
]

class SimpleBagRecorder(Node):
    def __init__(self,topics):
        super().__init__('simple_bag_recorder')
        self.callbacks = []
        self.subscriptions_list = []
        self.topic_infos = []
        self.writer = rosbag2_py.SequentialWriter()
        self.topics = topics
        storage_options = rosbag2_py._storage.StorageOptions(
            uri=datetime.datetime.now().strftime('rosbag2_%Y_%m_%d-%H_%M_%S'),
            storage_id='sqlite3')
        converter_options = rosbag2_py._storage.ConverterOptions('', '')
        self.writer.open(storage_options, converter_options)

        i = 0
        for topic in self.topics:
            self.topic_infos.append(rosbag2_py._storage.TopicMetadata(
                name=topic[0],
                type=topic[1],
                serialization_format='cdr'))
            self.writer.create_topic(self.topic_infos[i])
            self.subscriptions_list.append( self.create_subscription(
                topic[2],
                topic[0],
                lambda msg:  self.topic_callback(msg, topic[0]),
                10))
            i+=1
    def topic_callback(self, msg,  topic):
        print('Called : ',topic)
        self.writer.write(
            topic,
            serialize_message(msg),
            self.get_clock().now().nanoseconds)

def main(args=None):
    rclpy.init(args=args)
    sbr = SimpleBagRecorder(topics)
    for sub in sbr.subscriptions:
        print (sub.topic)
    rclpy.spin(sbr)
    rclpy.shutdown()

if __name__ == '__main__':
    main()`

fujitatomoya, 2024-01-22:
i guess the same lambda signature with const `topic[0]` is registered for the callbacks in the all subscriptions.

see https://github.com/fujitatomoya/ros2_test_prover/blob/5be23231f1c49583171c2504a33d97740b99ab3d/prover_rclpy/src/rosbag2_1549.py#L40, that can solve your problem.

It is highly recommended to ask the questions for https://robotics.stackexchange.com/

D-2n, 2024-01-23:
> i guess the same lambda signature with const `topic[0]` is registered for the callbacks in the all subscriptions.
>
> see [...]
Great, thank you! It works now.

closed 2024-01-23 by D-2n