ros2/rosbag2 #1554: Add play --exclude option for humble

tmxkn1, 2024-01-29:
## Description

`play --exclude` has been available on `rolling` branch since 2 years ago but still not pulled to `humble`.

## Completion Criteria

`play --exclude` is added for `humble` branch.

## Implementation Notes / Suggestions

I can work on this feature but would like to know if there is a reason it has not yet been done.

MichaelOrlov, 2024-01-29:
The main reason is that we need to keep API and ABI compatibility for branches corresponding to the ROS2 distros, for Humble in this case too.
If you can make it without breaking API and ABI compatibility you are more than welcome to do it.

MichaelOrlov, 2024-02-23:
To add `play --exclude` option to Humble will need to backport https://github.com/ros2/rosbag2/issues/1046.
However, there are required to add a new field to the `struct StorageFilter::topics_regex_to_exclude` from storage_filter.hpp and  to the `struct PlayOptions::topics_regex_to_exclude` from play_options.hpp as part of the #1046.

According to the https://acodersjourney.com/20-abi-breaking-changes/ 
> 4. Add, remove, or re-order member variables in a class

These changes are considered as ABI-breaking changes. 
- Therefore closing this issue as will not be done.

closed 2024-02-23 by MichaelOrlov