ros2/rosbag2 #1559: ros-humble-zstd-vendor have old version for libzstd library and this is used by systemd

ionutnechita-intel, 2024-01-31:
ros-humble-zstd-vendor have old version for libzstd library

## System (please complete the following information)
 - OS: Ubuntu Jammy
 - ROS 2 Distro: Humble
 - Install Method apt
 - Version: 0.15.9-1jammy.20240125.201047

First issue:
ros-humble-zstd-vendor have 1.4.4 version, but Ubuntu have 1.4.8.

ROS:
/opt/ros/humble/lib/pkgconfig/libzstd.pc
/opt/ros/humble/lib/libzstd.so.1.4.4
/opt/ros/humble/lib/libzstd.so
/opt/ros/humble/lib/libzstd.so.1

Ubuntu:
/usr/lib/x86_64-linux-gnu/pkgconfig/libzstd.pc
/usr/lib/x86_64-linux-gnu/libzstd.so.1
/usr/lib/x86_64-linux-gnu/libzstd.so.1.4.8
/usr/lib/x86_64-linux-gnu/libzstd.a
/usr/lib/x86_64-linux-gnu/libzstd.so

And i notice other issue.
Second issue:

systemd binary use this library from ROS location.

**_libzstd.so.1 => /opt/ros/humble/lib/libzstd.so.1 (0x00007f72de49f000)_**

```
ldd /usr/bin/systemd
        linux-vdso.so.1 (0x00007ffe4d26b000)
        libsystemd-shared-249.so => /lib/systemd/libsystemd-shared-249.so (0x00007f72deeb6000)
        libseccomp.so.2 => /lib/x86_64-linux-gnu/libseccomp.so.2 (0x00007f72dee6f000)
        libselinux.so.1 => /lib/x86_64-linux-gnu/libselinux.so.1 (0x00007f72dee43000)
        libmount.so.1 => /lib/x86_64-linux-gnu/libmount.so.1 (0x00007f72dedff000)
        libpam.so.0 => /lib/x86_64-linux-gnu/libpam.so.0 (0x00007f72deded000)
        libaudit.so.1 => /lib/x86_64-linux-gnu/libaudit.so.1 (0x00007f72dedbd000)
        libkmod.so.2 => /lib/x86_64-linux-gnu/libkmod.so.2 (0x00007f72deda1000)
        libapparmor.so.1 => /lib/x86_64-linux-gnu/libapparmor.so.1 (0x00007f72ded8c000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f72deb63000)
        libacl.so.1 => /lib/x86_64-linux-gnu/libacl.so.1 (0x00007f72deb59000)
        libblkid.so.1 => /lib/x86_64-linux-gnu/libblkid.so.1 (0x00007f72deb22000)
        libcap.so.2 => /lib/x86_64-linux-gnu/libcap.so.2 (0x00007f72deb15000)
        libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007f72deadb000)
        libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007f72de99d000)
        libip4tc.so.2 => /lib/x86_64-linux-gnu/libip4tc.so.2 (0x00007f72de993000)
        liblz4.so.1 => /lib/x86_64-linux-gnu/liblz4.so.1 (0x00007f72de973000)
        libcrypto.so.3 => /lib/x86_64-linux-gnu/libcrypto.so.3 (0x00007f72de530000)
        libzstd.so.1 => /opt/ros/humble/lib/libzstd.so.1 (0x00007f72de49f000)
        liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f72de474000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f72df336000)
        libpcre2-8.so.0 => /lib/x86_64-linux-gnu/libpcre2-8.so.0 (0x00007f72de3dd000)
        libcap-ng.so.0 => /lib/x86_64-linux-gnu/libcap-ng.so.0 (0x00007f72de3d5000)
        libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f72de3af000)

```

clalancette, 2024-01-31:
Yeah, this is a bug.  We shouldn't unconditionally vendor zstd here; we should only do that for platforms where we don't have zstd available.  I'll look at submitting a PR for this to Rolling, which we can then consider backporting to Humble and Iron.

clalancette, 2024-01-31:
Oh, actually.  This is a bug only in Humble; it was fixed in Iron (and later) by https://github.com/ros2/rosbag2/pull/1111 .  We just need to backport that one.

ionutnechita-intel, 2024-02-06:
Backport to humble is merged now.

Thanks.

PR #1560 (2024-02-01): Fix/zstd vendor does not find system zstd (backport #1111)
This is an automatic backport of pull request #1111 done by [Mergify](https://mergify.com).

---

<details>
<summary>Mergify commands and options</summary>

<br />

More conditions and actions can be found in the [documentation](https://docs.mergify.com/).

You can also trigger Mergify actions by commenting on this pull request:

- `@Mergifyio refresh` will re-evaluate the rules
- `@Mergifyio rebase` will rebase this PR on its base branch
- `@Mergifyio update` will merge the base branch into this PR
- `@Mergifyio backport <destination> [... cut]
closed 2024-02-06 by ionutnechita-intel