ros2/rosbag2 #1572: Failed to seek rosbag2

atinfinity, 2024-02-26:
## Description

I tried to use seek API of `rosbag2_py`.
But, I could not seek rosbag2.

```py
import rosbag2_py

reader = rosbag2_py.SequentialReader()
input_bag = "rosbag2/rosbag2_py/test/resources/talker/talker.db3"
reader.open(
    rosbag2_py.StorageOptions(uri=input_bag, storage_id="sqlite3"),
    rosbag2_py.ConverterOptions(
        input_serialization_format="cdr", output_serialization_format="cdr"
    ),
)

reader.seek(100)
topic1, data1, timestamp1 = reader.read_next()
print(f"timestamp1 = {timestamp1}")

reader.seek(0)
topic2, data2, timestamp2 = reader.read_next()
print(f"timestamp2 = {timestamp2}")
```

### Expected Behavior

```py
reader.seek(100)
topic1, data1, timestamp1 = reader.read_next()
print(f"timestamp1 = {timestamp1}")

reader.seek(0)
topic2, data2, timestamp2 = reader.read_next()
print(f"timestamp2 = {timestamp2}")
```

I expected that `timestamp1` and `timestamp2` are different value.

### Actual Behavior

```py
reader.seek(100)
topic1, data1, timestamp1 = reader.read_next()
print(f"timestamp1 = {timestamp1}")

reader.seek(0)
topic2, data2, timestamp2 = reader.read_next()
print(f"timestamp2 = {timestamp2}")
```

But `timestamp1` and `timestamp2` is same value.

## To Reproduce

```shell
sudo apt install ros-humble-rosbag2-py
git clone https://github.com/ros2/rosbag2.git -b humble
```

And, I run the following script.

```py
import rosbag2_py

reader = rosbag2_py.SequentialReader()
input_bag = "rosbag2/rosbag2_py/test/resources/talker/talker.db3"
reader.open(
    rosbag2_py.StorageOptions(uri=input_bag, storage_id="sqlite3"),
    rosbag2_py.ConverterOptions(
        input_serialization_format="cdr", output_serialization_format="cdr"
    ),
)

reader.seek(100)
topic, data, timestamp = reader.read_next()
print(f"timestamp = {timestamp}")

reader.seek(0)
topic, data, timestamp = reader.read_next()
print(f"timestamp = {timestamp}")
```

And, I used [rosbag2_py/test/resources/talker/talker.db3](https://github.com/ros2/rosbag2/blob/humble/rosbag2_py/test/resources/talker/talker.db3) for this check.

## System (please complete the following information)

 - OS: Ubuntu 22.04
 - ROS 2 Distro: ROS 2 Humble
 - Install Method: APT
 - Version: release

## Additional context

I also checked on ROS 2 Iron and got the same result.

dskkato, 2024-02-26:
As you confirmed the start/end times of this bag file using the "ros2 bag info" command, it is  approximately 4.5 seconds records from 1585866235.1 [sec] to 1585866239.6 [sec]. Since the seek function accepts timestamps in nanosecond units, if you set start_timestamp_sec = 1585866236, you should get results as if seeking up to that point.

```py
import rosbag2_py

reader = rosbag2_py.SequentialReader()
input_bag = "rosbag2/rosbag2_py/test/resources/talker/talker.db3"
reader.open(
    rosbag2_py.StorageOptions(uri=input_bag, storage_id="sqlite3"),
    rosbag2_py.ConverterOptions(
        input_serialization_format="cdr", output_serialization_format="cdr"
    ),
)
start_timestamp_sec = 1585866236
reader.seek(start_timestamp_sec * 10 ** 9)
topic1, data1, timestamp1 = reader.read_next()
print(f"timestamp1 = {timestamp1}")

reader.seek(0)
topic2, data2, timestamp2 = reader.read_next()
print(f"timestamp2 = {timestamp2}")
```

output

```sh
timestamp1 = 1585866236112742168
timestamp2 = 1585866235112411371
```

atinfinity, 2024-02-26:
@dskkato Thank you for your comment. I understood.

> start_timestamp_sec = 1585866236

I found that this value could get from metadata.

```py
metadata = rosbag2_py.Info().read_metadata(input_bag, "")
print(f"starting_time = {metadata.starting_time.timestamp()} sec.")
```

### output

```
starting_time = 1585866235.112411 sec.
```

MichaelOrlov, 2024-02-26:
@dskkato Thanks for your support on this issue.
- Closing this issue as it is not a bug and behavior consistent with design and API spec.

closed 2024-02-26 by MichaelOrlov