ros2/rosbag2 #1573: Docker- No storage id specified, and no plugin found that could open URI

adijams01, 2024-02-26:
## Description
I am trying to play rosbag on docker
I ran the launch file for lidar it worked, but while playing the rosbag i am getting error

### Expected Behavior
it should have been open and viewed on rviz2

### Actual Behavior
```
root@fbae67ab16dd:~/ros2_ws# ros2 bag play /my_src/ros2_ws/src/dataset/walking_dataset.bag

closing.

closing.
[ERROR] [1708928873.483955608] [rosbag2_storage]: No storage id specified, and no plugin found that could open URI
No storage could be initialized from the inputs.
```

## To Reproduce
[link](https://github.com/TixiaoShan/LIO-SAM/blob/ros2/Dockerfile) to dockerfile
[link](https://drive.google.com/file/d/1fpoOPVxbQbFmmLd2Pe2u8LnYueZLGtOm/view?usp=drive_link) to walking dataset's rosbag
```
docker run --name Lidar_r2 -it -v $PWD/src:/my_src -v /tmp/.X11-unix:/tmp/.X11-unix:rw --env=DISPLAY 86d3c223e99b bash
docker container start -i Lidar_r2
```
once in container 
on first terminal use launch 
second terminal inside container  play rosbag 

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: Foxy (humble in docker)

Thank you

MichaelOrlov, 2024-02-26:
@adijams01 Can you please try to run 
```
ros2 bag list storage --verbose
```
in your Docker.
What output do you see?

Spirquel, 2024-02-27:
it's because you use a file with ".bag" and these files is for ROS1 (the ros for students, university and search). So for ROS2 the rosbag2 file is ".db3". it exists others files like ".zstd" which is a compressor (like a .zip) for a files rosbag.
".mcap" and ".sqlite3" i don't very understand what it is but if it interest you, the README talk about that.
If you want to convert your .bag the answer is [here](https://github.com/ros2/rosbag2_bag_v2/blob/master/README.md) but you have to install [ROS1 Distribution](http://wiki.ros.org/Distributions) and [ROS2 Distribution](https://docs.ros.org/en/humble/Releases.html)

adijams01, 2024-02-28:
Thank you, for your responses now i am able to view it in rviz

closed 2024-02-28 by adijams01

Spirquel, 2024-03-25:
I have a good new. If you want to convert .bag to .db3, it exist a python code with like name "rosbags" the link is [here](https://pypi.org/project/rosbags/) and for that you needn't to install ros1 distribution.