ros2/rosbag2 #1574: Record a limited number of messages on each topic

AadityaRavindran, 2024-02-27:
## Description

rosbag record has an argument which records a limited number of messages of any topic(s). Eg: `rosbag record -l 1000 /chatter`

## Related Issues

## Completion Criteria

- Add an option `ros2 bag record --limit <NUM> <topics>`

## Implementation Notes / Suggestions

## Testing Notes / Suggestions
- Recording should stop after receiving NUM messages on each topic

MichaelOrlov, 2024-02-28:
@AadityaRavindran But how to count the number of messages to record if multiple topics need to be recorded with different frequencies?

fujitatomoya, 2024-02-28:
@AadityaRavindran besides ROS 1 rosbag supports it, what use case do you have? i think this is reasonable option to have, but what could be the actual use case?

AadityaRavindran, 2024-02-28:
> @AadityaRavindran besides ROS 1 rosbag supports it, what use case do you have? i think this is reasonable option to have, but what could be the actual use case?

@fujitatomoya I have integration tests where I need to record 1 message each of a few topics (Eg: the static map). These messages are used as inputs to gtest a class. I wrote my own bag recorder using rosbag2_cpp, but it would be extremely useful if I could generalize this to other gtests. I could write a generic bag recorder which takes in just 1 message of a list of topics, but if it was baked into rosbag2 cli, it would be helpful.

In general, if you have topic interfaces to a class, it's extremely useful to be able to write a unit/integration test (ros2 lifecycle nodes makes this easy to write and fast to run)
1. which reads in the collected bag file using rosbag2_cpp api
2. publish the data needed by the test through lifecycle nodes
3. test your use cases

If there were multiple messages needed at different times or order, you would store 1 message each of the various topics in different bags, and play them back whenever needed by the test.