ros2/rosbag2 #1581: Read topics from config file when recording

tonynajjar, 2024-03-11:
## Description

I'd like to be able to able to run `ros2 bag record --topics-config-file topics-config-file.yaml` (or similar) where `topics-config-file.yaml` would contain a list of topics to record. The main benefit of this approach would be to have a config file that is version-controllable, e.g. at some site we want to record different topics.

## Related Issues

None that I could find

## Completion Criteria

** What needs to be true before we can call this "Done"? Bullet lists are appropriate. **

## Implementation Notes / Suggestions

** If you have ideas about how this feature might be accomplished, put them here. Note that this is just a suggestion to the implementer, so feel free to speculate. **

## Testing Notes / Suggestions

** All features in this project need tests. Please give some input on cases that will need to be tested - and how the testing might be implemented. **

fujitatomoya, 2024-03-11:
@tonynajjar i think this is useful for supporting 3rd party application when we do debugging.

besides a list of `topics` and `services`, maybe the following could be also considered in the configuration file?

- `ros2 bag record` options can be also described in the configuration file?
- QoS override setting to each topic or service? (i think currently this is not supported yet in rosbag2.)

tonynajjar, 2024-03-11:
> QoS override setting to each topic or service? (i think currently this is not supported yet in rosbag2.)

@fujitatomoya this I believe can be done already:
https://docs.ros.org/en/foxy/How-To-Guides/Overriding-QoS-Policies-For-Recording-And-Playback.html

MichaelOrlov, 2024-03-20:
@tonynajjar @fujitatomoya IMO the requested functionality already exists and could be achieved through composition via starting reorder from the launch file. Sorry the documentation about it is still in review. See https://github.com/ros2/rosbag2/pull/1510/files#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5R349-R365 and you are more than welcome with a contribution to improve our documentation in this regards.

The short example would be:
One instead of running command `ros2 bag record --record-config-file config-file.yaml` can run
```sh
$ ros2 launch rosbag2_transport record_all.launch.py
```
where the `record_all.launch.py` could be
```python
import launch

def generate_launch_description():
    return launch.LaunchDescription([
        launch.actions.Node(
            package='rosbag2_transport',
            executable='recorder',
            name='rosbag2_recorder',
            output="screen",
            parameters=["/path/to/recorder_params.yaml"],
        )
    ])
```
Example of the `recorder_params.yaml` can be found here https://github.com/ros2/rosbag2/blob/rolling/rosbag2_transport/test/resources/recorder_node_params.yaml

tonynajjar, 2024-03-20:
Thanks @MichaelOrlov for the hint, looks indeed promising. I will test it out and get back to you (can't promise when but likely before end of next week)

tonynajjar, 2024-03-21:
Actually it looks like I won't be able to test it since I'm on Humble and the rosbag2_recorder  isn't available there. Any plans to backport it?
In any case, if you say it's supported on rolling then feel free to close the issue and I'll report back here if I get to use it and face issues

MichaelOrlov, 2024-03-21:
@tonynajjar Backporting is not feasible due to the massive changes that are not API/ABI compatible.
Feedback will be useful. 
I am also considering if we could add `-record-config-file` CLI parameter with a low effort via parsing yaml file as a node parameters on the rosbag2_py layer.

RMichaelSwan, 2024-10-14:
Is there any existing way to configure a recorder to record multiple sets of topics or is that better done by running multiple recorders in parallel (i.e. #1548 )? The use-case would be to separate high bandwidth topics from low bandwidth topics which would aid in debugging (don't have to share large bags all the time if less data is needed to replicate).

MichaelOrlov, 2024-10-14:
You can use multiple instances of the Rosbag2 recorder running in parallel or use `ros2 bag convert` as a post-processing tool to split one large bag into multiple smaller bags with a different subset of topics.
See details in the https://github.com/ros2/rosbag2?tab=readme-ov-file#converting-bags
```
Example split:

$ ros2 bag convert -i bag1 -o out.yaml

# out.yaml
output_bags:
- uri: split1
  topics: [/topic1, /topic2]
- uri: split2
  topics: [/topic3]
```