ros2/rosbag2 #1595: Specify bags directory

tonynajjar, 2024-03-25:
## Description

Currently in `record.py`, the arg `--output` allows to specify the full directory of the bag. It would be nice if we could specify the "base directory" only such that the bag folder is still automatically generated with the timestamp but all bags are contained in a configurable directory other than the current directory where the command is run

fujitatomoya, 2024-03-25:
i think sounds reasonable, `--output-dir` or environmental variable would be useful.

MichaelOrlov, 2024-03-26:
@tonynajjar I am sorry. I didn't get your proposal. Could you please clarify/elaborate?

tonynajjar, 2024-03-26:
Sure. Currently we can specify the output dir which is base directory + bag folder. If we don't specify it then base dir = current dir and bag folder is autogenerated to be unique (with a timestamp).

I'd like to specify only the base dir as a ln argument, to keep both benefits of specifying a directory for my bags and have unique aurogenerated names. I hope it's clearer

MichaelOrlov, 2024-03-26:
You basically want the autogenerated name of the `bag_folder` when specifying a base path where it should be?

Make sense to consider changing the current behaviour for the `--output` flag? 
What would be the pros and cons?

I am not a fan of having two very similar options like `--output` and `--output-dir`. It will cause a lot of confusion and questions. Why and what is the difference?

tonynajjar, 2024-03-26:
> Make sense to consider changing the current behaviour for the --output flag? What would be the pros and cons?

The biggest con is probably that it's breaking for current users. Besides, I think it makes sense to want to specify these 2 things separately: where you want to save your bags and what you want the name of your bag to be

MichaelOrlov, 2024-05-05:
@tonynajjar In this case, to avoid confusion in parameter names, I would suggest adding a new parameter like `--autogen-name` instead of the `--output-dir.` 
i.e., one would use `ros2 bag record --output path/to/bag/folder --autogen-name` if needed to autogenerate the folder name inside the specified base folder path in the `--output` argument.

b-adkins, 2024-08-23:
I don't think it's a stretch to say that robotics projects want ros bag to launch automatically as part of their stack bringup, aka in a launch file. The problem with launch files and the `ros2 bag` interface is how they can be run from a working directory anywhere. So your logs will be scattered all over the place.

I also didn't want to have to write my own name generators. But to change the directory, you are required to.

Here's my workaround:

1. Use the -o flag
2. Copy paste the filename-generating python code into the `launch.py` file, from `ros2bag/verb/record.py:293`
```
bag_dirname = datetime.datetime.now().strftime('rosbag2_%Y_%m_%d-%H_%M_%S')
```
3. Do some path stuff:
```
ExecuteProcess(
        cmd=['ros2', 'bag', 'record', '-a', '-o', 'os.path.join("my_bag_root", bag_dirname)']
...
```
(If you're not using launch.py, then I guess you can be sad? Or wrap it in a shell script?)

emersonknapp, 2024-08-27:
I'm not sure I see that it's necessary for `rosbag2` to support this behavior. You can easily set the `cwd` for the recording process that you launch, no matter what environment you're in, no?

1. When running from shell: `pushd $MY_BASE_DIR && ros2 bag record ...`
2. In a launchfile:

```
ExecuteProcess(
  cmd=['ros2', 'bag', 'record', ...],
  cwd=my_base_dir,  
),
```

Or have I missed something about this request that changing `cwd` doesn't address?

emersonknapp, 2025-02-21:
Resolving conversation as stale given no reply.

I'm going to close this issue as `wontfix` because the behavior can be trivially accomplished by the calling environment, so `rosbag2` itself does not need to overcomplicate things by adding a new feature.

For future reference, note the solution from my previous comment:

> I'm not sure I see that it's necessary for `rosbag2` to support this behavior. You can easily set the `cwd` for the recording process that you launch, no matter your invoking enironment, no?
> 
> 1. When running from shell: `pushd $MY_BASE_DIR && ros2 bag record ...`
> 2. In a launchfile:
> 
> ```
> ExecuteProcess(
>   cmd=['ros2', 'bag', 'record', ...],
>   cwd=my_base_dir,  
> ),
> ```
> 
> Or have I missed something about this request that changing `cwd` doesn't address?

closed 2025-02-21 by emersonknapp