ros2/rosbag2 #1602: wrong duration and missing topics/frames after ros2 bag convert

maxklock, 2024-04-09:
## Description
I have a bagfile and wanted to split it into several files using `ros2 bag convert`, but the resulting bagfile is not working as expected.

### Expected Behavior
I have the original bagfile with a single file of nearly 4GB. With convert I want to split this file into ~500MB parts.
Both, the old and the new bagfile, should behave the same.

### Actual Behavior
When I use `ros2 bag play` on the old bagfile, everything works fine. But when using the new bagfile, i get a lot missing frames and topics.
When inspecting both files with `ros2 bag info` the topic information and message count is the same, but the duration and start time differs.

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Run `ros2 bag play rosbag_origin` to validate you have a working bagfile. No erros in the logs and the process in running ~80s.
2. Run `ros2 bag convert` to modify the bagfile:
```
ros2 bag convert -i rosbag_origin -o convert.yml
[INFO] [1712645045.805924828] [rosbag2_storage]: Opened database 'rosbag_origin/rosbag_origin_0.db3' for READ_ONLY.
[INFO] [1712645045.810819109] [rosbag2_storage]: Opened database 'rosbag_convert/rosbag_convert_0.db3' for READ_WRITE.
[INFO] [1712645047.498534912] [rosbag2_storage]: Opened database 'rosbag_convert/rosbag_convert_1.db3' for READ_WRITE.
[INFO] [1712645049.129373085] [rosbag2_storage]: Opened database 'rosbag_convert/rosbag_convert_2.db3' for READ_WRITE.
[INFO] [1712645050.516206986] [rosbag2_storage]: Opened database 'rosbag_convert/rosbag_convert_3.db3' for READ_WRITE.
[INFO] [1712645051.811610194] [rosbag2_storage]: Opened database 'rosbag_convert/rosbag_convert_4.db3' for READ_WRITE.
[INFO] [1712645053.199245268] [rosbag2_storage]: Opened database 'rosbag_convert/rosbag_convert_5.db3' for READ_WRITE.
[INFO] [1712645054.700044665] [rosbag2_storage]: Opened database 'rosbag_convert/rosbag_convert_6.db3' for READ_WRITE.
[INFO] [1712645056.326892823] [rosbag2_storage]: Opened database 'rosbag_convert/rosbag_convert_7.db3' for READ_WRITE.
[INFO] [1712645058.296396332] [rosbag2_storage]: Opened database 'rosbag_convert/rosbag_convert_8.db3' for READ_WRITE.
```
with convert.yml
```
output_bags:
  - uri: rosbag_convert
    all: true
    max_bagfile_size: 500000000
```
3. Run `ros2 bag play rosbag_convert`, which will end after ~9s. Furthermore, the individual files open within milliseconds.
4. Run `ros2 bag info` on both files to see different durations and times:
```
ros2 bag info rosbag_origin
  Files:             rosbag_origin_0.db3
  Bag size:          3.8 GiB
  Storage id:        sqlite3
  Duration:          80.229s
  Start:             Apr  8 2024 10:47:45.72 (1712566065.72)
  End:               Apr  8 2024 10:49:05.301 (1712566145.301)
  Messages:          49329
```
```
ros2 bag info rosbag_convert
  Files:             rosbag_convert_0.db3
                     rosbag_convert_1.db3
                     rosbag_convert_2.db3
                     rosbag_convert_3.db3
                     rosbag_convert_4.db3
                     rosbag_convert_5.db3
                     rosbag_convert_6.db3
                     rosbag_convert_7.db3
                     rosbag_convert_8.db3
  Bag size:          3.8 GiB
  Storage id:        sqlite3
  Duration:          9.895s
  Start:             Apr  8 2024 10:49:03.894 (1712566143.894)
  End:               Apr  8 2024 10:49:13.790 (1712566153.790)
  Messages:          49329
```

## System (please complete the following information)
 - OS: Ubuntu 22.04
 - ROS 2 Distro: Humble
 - Install Method: APT
 - Version: 0.15.9

## Additional context
I've changed the file names in the output - if there are inconsistent names in the output, I'm sorry.
I used `apt upgrade` to install the latest ros2bag version

maxklock, 2024-04-09:
I found a manual workaround to make the bagfile working:
- open `metadata.yaml` of rosbag_convert
- change `duration` and `starting_time` to match the values of rosbag_origin

Afterwards the bagfile is working as expected

maxklock, 2024-04-16:
The same issure occures when using max-bag-size during recording. My bagfiles are not working out of the box, due to wrong duration and starting_time. In contrast to the convert, however, I have no reference data to correct the metadata.yaml manually.

MichaelOrlov, 2024-04-20:
- @maxklock This issue is likely related to the https://github.com/ros2/rosbag2/pull/1616

MichaelOrlov, 2024-05-05:
- Closing as fixed in the #1616. Feel free to reopen if the issue still reproduces.

closed 2024-05-05 by MichaelOrlov