ros2/rosbag2 #1604: Won't play parallel bags in two different folders

michael-code19, 2024-04-09:
## Description
I have multi-file bag files in two folders, /folder1/folder1_sub/ and /folder2/folder2_sub/. I want to play back both the folders at the same time. Using the folder directory ros2 bag play folder1/folder1_sub folder2/folder2_sub does not play the bag files in parallel.  

EX I want this to work:

ros2 bag play LIDAR_TOPICS/LIDAR_TOPICS OTHER_TOPICS/other_topics

### Expected Behavior
I expect ros2 bag play to open both folders and play the bags in parallel

### Actual Behavior
usage: ros2 [-h] [--use-python-default-buffering] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: unrecognized arguments: OTHER_TOPICS/other_topics

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Create two multi-file bag files and put them in folder1/folder1_sub/ and folder2/folder2_sub
2. Attempt to play them back in parallel using ros2 bag play folder1/folder1_sub/ and folder2/folder2_sub
3. You will get error

## System (please complete the following information)
 - OS: Ubuntu
 - ROS 2 Distro: Humble
 - Install Method: APT
 - Version: 0.15.9

## Additional context
** Add any other context about the problem here **

fujitatomoya, 2024-04-17:
how about merging bag files into one via `ros2 bag convert -i xxx -i yyy -o zzz.yaml` and the use it with `ros2 bag play zzz`? see more details for https://github.com/ros2/rosbag2?tab=readme-ov-file#converting-bags

MichaelOrlov, 2024-04-19:
@michael-code19 
> Using the folder directory ros2 bag play folder1/folder1_sub folder2/folder2_sub does not play the bag files in parallel.
> EX I want this to work:
> ros2 bag play LIDAR_TOPICS/LIDAR_TOPICS OTHER_TOPICS/other_topics

To accomplish your wants, you need to understand that someone behind the scenes should run multiple instances of the rosbag2 player for each bag file which is supposed to be playing in parallel. 
If we do so, then someone could complain that he wants or expects to be able to provide different filters or other configurations for each played bag file or complain that one bag started playing a few milliseconds earlier than the other or the player runs out of memory or CPU resources or something else.
We don't want to mess up and we will not go this way!!!
The current behavior is a WAD (Works As Designed).
However, we allow one to do it themselves.
In the world of recording and playback, this feature names distributed recording or playback.
If one wants to playback multiple bags in parallel simultaneously, he can start multiple instances of the player in pause mode first. i.e., start each player instance in a separate terminal window or process with the CLI option `--start-paused`.
Then resume playback on all instances via one service call 
```bash
ros2 service call rosbag2_player/resume rosbag2_interfaces/Resume
```

MichaelOrlov, 2024-04-19:
- Closing since this is not a bug and rosbag2 works as designed.

closed 2024-04-19 by MichaelOrlov

danil-tolkachev, 2024-07-03:
> this is not a bug and rosbag2 works as designed

@MichaelOrlov sorry to hear that. In ROS1 I was [able to play multiple bag-files](http://wiki.ros.org/rosbag/Commandline#play) at the same time.