ros2/rosbag2 #1619: When recording a large rosbag, metadata file shoud be written incrementally

randaz81, 2024-04-19:
## Description

Let's consider a use case in which the recording of a rosbag lasts several days. When using the split option, the data file is correctly split into multiple files every hour, but no metadata file is written until the process successfully terminates. If a blackout occurs, no metadata is written at all, and the whole data acquisition becomes unusable.

## Related Issues

--

## Completion Criteria

--

## Implementation Notes / Suggestions

The metadata file should be written and updated when the data split occurs.

## Testing Notes / Suggestions

--

MichaelOrlov, 2024-04-20:
Hi @randaz81 I don't think that this is a problem on the current Rolling.  
Could you please add more information about how to reproduce the issue described?

For reference, we currently keep metadata file only for backward compatibility and to be able to edit it and see its content in the text editor for convenience. We also writing metadata to each bag file directly twice. Once on file creation and the second time on file closing.

MichaelOrlov, 2024-05-05:
@randaz81 BTW if after recording you are missing `metadata.yaml` file you can easily recreate it with 
`ros2 bag reindex path/to/folder/with/recorded/bags`

MichaelOrlov, 2024-05-05:
- Closing as WAD (Works As Designed)

closed 2024-05-05 by MichaelOrlov