ros2/rosbag2 #1621: Frame loss occurs when ROS2 plays point cloud data on Intel 13th generation CPU

DMUZhangXianglong, 2024-04-22:
Recently I posted a question like this. Install ROS2 humble on Intel 13th generation CPU i9-13900HX i7-13700KF i5-13600KF, then play ros2 bag, open rviz2 to view point cloud data, as shown below, when Best Effort is selected, There will be point cloud frame loss, which is manifested as screen freezes. Use ros2 topic hz to view the point cloud topic, the display is much lower than 10hz

In the C++ code implementation, if it is RMW QOS POLICY RELIABILITY BEST EFFORT in the afternoon, the frequency of subscribing to point clouds is very low. When it is changed to RMW QOS POLICY RELIABILITY RELIABLE, it is normal.

The above phenomenon occurs on all three devices. The system is Ubuntu22.04.4 and the kernel version is 6.5.

I don’t know if you have the above phenomenon in the ubuntu22.04 ros2 humble environment installed on the 13th generation intel CPU.
[image]
[image]

MichaelOrlov, 2024-04-22:
@DMUZhangXianglong Sorry, but how is it related to the rosbag2?
The QoS reliability best effort assumes that messages could be lost it looks like a WAD (Works as Designed).

MichaelOrlov, 2024-06-26:
- Closing as not an issue. The rosbag2 behaves as designed.

closed 2024-06-26 by MichaelOrlov