ros2/rosbag2 #1623: ros2 bag record -a --compression-mode message --compression-format zstd can't play

LHZ-code, 2024-04-24:
use ros2 bag record -a --compression-mode message --compression-format zstd get a .db3. But the .db3 can not ros2 bag play .u Use ros2 bag play the .db3. get the issue ros2 topic echo /scan 
1713951203.062429 [99]       ros2: using network interface enp49s0 (udp/172.31.50.61) selected arbitrarily from: enp49s0, wlp0s20f3, docker0
1713951203.604090 [99]     recvUC: malformed packet received from vendor 1.16 state parse:data <52545053 02010110 db241001 99cbeaad 5f7e62c4 09010800 e3d12866 166f989a @0x20 15055009 00001000 00000000 03170000 00000000 fb000000 28b52ffd 607c1995> (note: maybe partially bswap'd) {{15,5,2384},0,16,0,1703,251}
1713951203.641438 [99]     recvUC: malformed packet received from vendor 1.16 state parse:data <52545053 02010110 db241001 99cbeaad 5f7e62c4 09010800 e3d12866 d77c25a4 @0x20 15055009 00001000 00000000 03170000 00000000 fc000000 28b52ffd 607c1985> (note: maybe partially bswap'd) {{15,5,2384},0,16,0,1703,252}
1713951203.677167 [99]     recvUC: malformed packet received from vendor 1.16 state parse:data <52545053 02010110 db241001 99cbeaad 5f7e62c4 09010800 e3d12866 becd48ad @0x20 15054c09 00001000 00000000 03170000 00000000 fd000000 28b52ffd 607c1965> (note: maybe partially bswap'd) {{15,5,2380},0,16,0,1703,253}

MichaelOrlov, 2024-05-09:
@LHZ-code Please use our [templates](https://github.com/ros2/rosbag2/tree/rolling/.github/ISSUE_TEMPLATE) for creating an issue. From what you have written, I have a hard time understanding what the problem is.

MichaelOrlov, 2024-05-16:
- Closing as a stale issue with no response from the author.

closed 2024-05-16 by MichaelOrlov

KaliyDmitriy, 2024-10-02:
i have a similar problem...