ros2/rosbag2 #1638: Rosbag2 does not work at all with CycloneDDS on Jazzy

clalancette, 2024-05-06:
## Description
During the Jazzy tutorial testing party, @mikaelarguedas found a severe issue where rosbag2 recording does not work at all with CycloneDDS.  The original report is in https://github.com/osrf/ros2_test_cases/issues/1279 , but in short when recording with `RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`, the timestamps are set to January 1, 1970.  The fallout from this is that playback never works.

### Expected Behavior
It should be possible to record and play rosbag2 data with `rmw_cyclonedds_cpp`

### Actual Behavior
No data recorded with `rmw_cyclonedds_cpp` can be played back.

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Terminal 1: `RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 launch composition composition_demo_launch.py`
2. Terminal 2: `RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 bag record -a`
3. Ctrl-C both Terminal 1 and Terminal 2
4. `ros2 bag info <rosbagfile>` - Notice that the start and end timestamps are January 1, 1970.
5. `RMW_IMPLEMENTATION ros2 bag play <rosbagfile>` - Nothing plays back

## System (please complete the following information)
 - OS: Ubuntu 24.04
 - ROS 2 Distro: Jazzy and Rolling
 - Install Method: apt
 - Version: 0.26.2-1noble.20240502

MichaelOrlov, 2024-05-06:
@clalancette I will take a look, likely relates to the https://github.com/ros2/rosbag2/pull/1531

clalancette, 2024-05-06:
> @clalancette I will take a look, likely relates to the #1531

Sounds good, thank you!

MichaelOrlov, 2024-05-07:
### RCA (Root Cause Analysis):
- in https://github.com/ros2/rosbag2/pull/1531, we switched to using "receive" timestamps from `message_info` during the recording by default.
- However, it turns out that the `CycloneDDS` doesn't support `received_timestamp` see https://github.com/ros2/rmw_cyclonedds/blob/c6dbe24b2f2be87cf8e4750d89501657ab83566f/rmw_cyclonedds_cpp/src/rmw_node.cpp#L3280-L3282
  ```
    message_info->source_timestamp = info.source_timestamp;
    // TODO(iluetkeb) add received timestamp, when implemented by Cyclone
    message_info->received_timestamp = 0;
  ```
   This is why timestamps for all messages with CycloneDDS are zeros.
- We do have unit test coverage for `send_timestamp` in the `test_record_all_use_sim_time.cpp`, but unfortunately, we missed updating generic recording integration tests to check recorded `received_timestamp`.