ros2/rosbag2 #1657: Interface of rosbag2_compression to know the topic type and use a different codec accordingly

facontidavide, 2024-05-15:
## Description

I would like to create a rosbag2_compression compression plugin that uses specialized codecs for images and pointclouds.

Unfortunately, looking at the [abstract interface](https://github.com/ros2/rosbag2/blob/rolling/rosbag2_compression/include/rosbag2_compression/base_compressor_interface.hpp), I can't see a way to determine the message type (I can just access the topic name.

Is there any way to work arounf that=

## Related Issues
Feature needed to create a specialized compression algorithm for pointclouds and depth images.

## Completion Criteria

Not sure... an additional virtual method that pass the topic type too?

https://github.com/ros2/rosbag2/blob/rolling/rosbag2_compression/include/rosbag2_compression/base_compressor_interface.hpp#L65-L67

## Implementation Notes / Suggestions

May be adding a method like this that I can override?

```cpp
  virtual void compress_serialized_bag_message(
    const rosbag2_storage::SerializedBagMessage * bag_message,
    rosbag2_storage::SerializedBagMessage * compressed_message,
    const std::string& topic_type) {
    compress_serialized_bag_message(bag_message, compressed_message);
}
```
## Testing Notes / Suggestions

TBD

MichaelOrlov, 2024-05-17:
Hi @facontidavide glad to see you at rosbag2 workspace.
I suggest discussing this topic at the biweekly Tooling Working Group meeting. The next one will be Tomorrow, Friday, at 9:00 a.m. Pacific Time US.
I encourage you to visit it. Here is the link on the ROS Events Calendar for the actual invite to add to your calendar: https://calendar.google.com/calendar/u/0/embed?src=agf3kajirket8khktupm9go748@group.calendar.google.com. 
Also direct link to the meeting https://www.google.com/url?q=https://meet.google.com/epe-iurw-hah?hs%3D122%26authuser%3D0&sa=D&source=calendar&ust=1716335329186245&usg=AOvVaw1aFBiQvuOkFnKYaCeY8Qb4
And link to the agenda doc https://docs.google.com/document/d/1Dsg_9XZQPhihpKQGQWMYTz2doGH4P2cAaNqr60cuNgw/edit?usp=meetingnotes&showmeetingnotespromo=true

TLDR: 
Even if we would provide a suggested API with `topic_type` and facilitate its implementation, there are other obstacles to implementing a compression plugin for dedicated types as you want.
1. Compressing and decompressing messages by given types will be very inefficient because Rosbag2 deals only with serialized data. You will need to deserialize data from the message, compress it as you like and then serialize it again in your plugin. The same procedure with deserialization, decompression and serialization again will be needed inside the plugin during playback.
2. The visualization tools like Foxglove Studio don't support per-message compression done via plugins in rosbag2 because they don't use rosbag2 API and libraries. Instead, Foxglove Studio supports compression methods built into the MCAP library. The rosbag2 also supports them. You can read more about it in the [recording-with-a-storage-configuration](https://github.com/ros2/rosbag2?tab=readme-ov-file#recording-with-a-storage-configuration) section of the rosbag2 readme file.
3. For the reasons mentioned in (2) and because of many user complaints, we disabled the ability to record with per-message compression from plugins in MCAP format.

cc: @defunctzombie

facontidavide, 2024-05-17:
Great points.

I know how to work around problem 1. efficiently.

Talking about problem 2, I am sure that a solution could exist, if foxglove includes the codec, but apparently they don't think that making smaller rosbags is an interesting problem to solve 😉

The challenge is that we should save in the MCAP schema some information about the compression method.

That is arguably complicated, but not impossible.

I guess that a easier solution is to use image_transport or pointcloud_transport instead, and record the compressed messages instead.

Thanks, I will close this for the time being

closed 2024-05-17 by facontidavide

defunctzombie, 2024-05-17:
> The challenge is that we should save in the MCAP schema some information about the compression method.

Probably point cloud messages need to evolve to include a compression format (similar to how CompressedImage messages have a format).

facontidavide, 2024-05-17:
We do have a low hanging fruit that would require the minimum amount of changes.

**pointcloud_transport** already has that: https://github.com/ros-perception/point_cloud_transport_plugins/blob/rolling/point_cloud_interfaces/msg/CompressedPointCloud2.msg

That is exactly the message definition that we need.

What I would suggest is to use `CompressedPointCloud2` into rosbag / MCAPs, instead of plain `PointCloud2`.

Then, it would be great if tools such as **RViz** and **Foxglove** could natively decompress those without any "middleman" involved.

But i am now convinced that this is not a change we can easily do at the level of rosbag itself