ros2/rosbag2 #1660: `ros2 bag convert` not writing any messages on jazzy

ottojo, 2024-05-17:
## Description
`ros2 bag convert` seems to not write any messages to the output bag on jazzy. I originally experienced this when trying to use the new parameters introduced in #1455, but the issue seems to occur with the "compress" example from the readme as well.

The output bag seems to not contain any data at all when running on jazzy.

### Expected Behavior
This is what happens on iron: The file size is reduced, but most importantly the resulting bag contains all topics and messages:

out.yaml (taken from readme):

```yaml
output_bags:
- uri: bag_compressed_iron
  all: true
  compression_mode: file
  compression_format: zstd
```

```console
jonas@ade ➜  ~ ros2 bag info bag_full_iron 

Files:             bag_full_iron_0.mcap
Bag size:          5.5 KiB
Storage id:        mcap
Duration:          33.999s
Start:             May 17 2024 19:14:06.45 (1715966046.45)
End:               May 17 2024 19:14:40.45 (1715966080.45)
Messages:          35
Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 35 | Serialization Format: cdr

jonas@ade ➜  ~ ros2 bag convert -i bag_full_iron -o out.yaml 
[INFO] [1715966134.703149096] [rosbag2_compression]: Compressing file: bag_compressed_iron/bag_compressed_iron_0.mcap

jonas@ade ➜  ~ ros2 bag info  bag_compressed_iron/

Files:             bag_compressed_iron_0.mcap.zstd
Bag size:          2.3 KiB
Storage id:        mcap
Duration:          33.999s
Start:             May 17 2024 19:14:06.45 (1715966046.45)
End:               May 17 2024 19:14:40.45 (1715966080.45)
Messages:          35
Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 35 | Serialization Format: cdr
```

### Actual Behavior
On jazzy, with out.yaml:
```yaml
output_bags:
- uri: compressed
  all: true
  compression_mode: file
  compression_format: zstd
```

The resulting file contains no messages at all:
```console
jonas@ade ➜  ~ ros2 bag info bag_jazzy_full 

Files:             bag_jazzy_full_0.mcap
Bag size:          6.8 KiB
Storage id:        mcap
ROS Distro:        jazzy
Duration:          24.999s
Start:             May 17 2024 19:16:57.716 (1715966217.716)
End:               May 17 2024 19:17:22.716 (1715966242.716)
Messages:          26
Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 26 | Serialization Format: cdr
Service:           0
Service information: 
jonas@ade ➜  ~ vim out.yaml
jonas@ade ➜  ~ ros2 bag convert -i bag_jazzy_full -o out.yaml 
[INFO] [1715966287.511906185] [rosbag2_compression]: Compressing file: compressed/compressed_0.mcap

jonas@ade ➜  ~ ros2 bag info compressed 

Files:             compressed_0.mcap.zstd
Bag size:          957 B
Storage id:        mcap
ROS Distro:        jazzy
Duration:          0.0s
Start:             Apr 12 2262 01:47:16.854 (9223372036.854)
End:               Apr 12 2262 01:47:16.854 (9223372036.854)
Messages:          0
Topic information: 
Service:           0
Service information: 

jonas@ade ➜  ~ ll compressed 
total 8.0K
-rw-r--r-- 1 jonas jonas 421 May 17 19:18 compressed_0.mcap.zstd
-rw-r--r-- 1 jonas jonas 536 May 17 19:18 metadata.yaml

jonas@ade ➜  ~ cat compressed/metadata.yaml 
rosbag2_bagfile_information:
  version: 9
  storage_identifier: mcap
  duration:
    nanoseconds: 0
  starting_time:
    nanoseconds_since_epoch: 9223372036854775807
  message_count: 0
  topics_with_message_count:
    []
  compression_format: zstd
  compression_mode: FILE
  relative_file_paths:
    - compressed_0.mcap.zstd
  files:
    - path: compressed_0.mcap
      starting_time:
        nanoseconds_since_epoch: 9223372036854775807
      duration:
        nanoseconds: 0
      message_count: 0
  custom_data: ~
  ros_distro: jazzy%
```

## To Reproduce
* Record a ros bag, such as a few seconds of `ros2 run demo_nodes_cpp talker` using the default settings `ros2 bag record -o bag_jazzy_full /chatter`
* Create an output config `out.yaml`, such as the compression example above taken from the readme
* convert the bag: `ros2 bag convert -i bag_jazzy_full -o out.yaml`
* inspect the output

## System
 - OS: Ubuntu 22.04, host `Linux ade 6.8.9-3-MANJARO`
 - ROS 2 Distro: Jazzy
 - Install Method: docker image based on `ros:jazzy` (APT, testing repositories)
 - Version:
```console
jonas@ade ➜  ~ apt-cache policy ros-jazzy-rosbag2
ros-jazzy-rosbag2:
  Installed: 0.26.3-1noble.20240516.143319
```

ottojo, 2024-05-19:
bisecting concluded " 3286736d364b2794051020ae8f2b0893521da16f is the first bad commit"

ottojo, 2024-05-19:
It seems like the above commit introduced a change in the config file format, namely removing the `all` and `exclude` key and adding the keys `all_topics, all_services, exclude_regex, exclude_topics, exclude_services`.

While i think that `all_topics` is a much better name than just `all` (i was wondering what that did and if it was required), this seems to break the examples in the readme and probably all existing config files out there as well.

I can submit a PR to fix the readme, but i think allowing the `all` keyword (as an alias for `all_topics` and `all_services`) would be useful as well...

MichaelOrlov, 2024-05-21:
@ottojo 
- The https://github.com/ros2/rosbag2/pull/1664 - should fix a problem with backward compatibility with old config files.

closed 2024-05-21 by MichaelOrlov