ros2/rosbag2 #1677: Feature request: rosbag2 reader that automatically detects if the bagfile is compressed or not

Rayman, 2024-05-29:
## Description

At the moment there are two readers:
1. `rosbag2_cpp::Reader`
2. `rosbag2_compression::SequentialCompressionReader`

If I use the normal reader on a compressed message bagfile I get this error:
```
'rmw_serialize: invalid data size, at ./src/rmw_node.cpp:1910'
```
If I use the compression reader on an uncompressed bagfile I get this
```
  what():  SequentialCompressionReader should not be initialized with NONE compression mode.
```
So that means I need to pick which reader based on the compression used in the bagfile.
But the bagfile contains the compression used, so there could be a reader that figures this out automatically.

## Related Issues

## Completion Criteria

A single reader interface that can read both compressed and uncompressed rosbags.

## Implementation Notes / Suggestions

## Testing Notes / Suggestions

MichaelOrlov, 2024-05-30:
Hi @Rayman, There is the only one robust way to detect whether the bag is compressed or not is to look into the `metadata.yaml` file and see what is inside the `compression_format` field.  
And we have already do this in the `ReaderWriterFactory::make_reader(storage_options)` api
https://github.com/ros2/rosbag2/blob/e01e1efe57c4f226eed21e2eefb0b459fd3bd0b8/rosbag2_transport/src/rosbag2_transport/reader_writer_factory.cpp#L25-L45

I would encourage you to use this factory method as well.

Rayman, 2024-05-30:
Ah nice, I was looking for something like that.

Is it a good idea to update the tutorials to make use of this API? I can open a PR for that

https://docs.ros.org/en/rolling/Tutorials/Advanced/Recording-A-Bag-From-Your-Own-Node-CPP.html
https://docs.ros.org/en/iron/Tutorials/Advanced/Reading-From-A-Bag-File-CPP.html

MichaelOrlov, 2024-05-30:
Hmm, for recording, we also have a factory method for creating the Writer. However, it relies on the recorder options.  
IMO, it will overcomplicate the tutorial.
Although for the https://docs.ros.org/en/iron/Tutorials/Advanced/Reading-From-A-Bag-File-CPP.html it does make sense to show how to use factory method. I would appreciate a PR for that.

Rayman, 2024-06-10:
I've updated the documentation, so I'll close this issue

closed 2024-06-10 by Rayman

Rayman, 2024-06-11:
Yes, it makes sense to have single `Reader` class that can be used in all circumstances (like in ROS1). And there could be an `UncompressedReader` and a `CompressedReader` underneath to do the actual work.