ros2/rosbag2 #1680: mcap_vendor files not found

CM134, 2024-05-30:
## Description

After upgrading ros-humble-mcap-vendor 0.15.11-1 on ARM I get the following error.
```sh
fatal error: mcap_vendor/mcap/writer.hpp: No such file or directory
   46 | #include <mcap_vendor/mcap/writer.hpp>
```
Before I was using 0.15.9-1, and my package compiled, but after the upgrade, it did not find the file. I can still see that the file exists in /opt/ros/humble/include/ and changing the include line in my code to have the full path to the file, my package builds.

I tested it also on amd64, and there it seems to find the files. 

## System
 - OS: Ubuntu 22.04.4 LTS on ARM
 - ROS 2 Distro: Humble
 - Install Method: APT
 - Version: 0.15.11-1jammy.20240522.165626

MichaelOrlov, 2024-07-16:
@CM134 Please clarify how are you building your code.
If building from sources please make sure that there are no leftovers from the previous build run.

MichaelOrlov, 2024-08-10:
- Closing as a stale issue with no response from the author.
- This is likely an issue with build environment on the user side because we didn't have any changes in the `mcap_vendor` package between mentioned `0.15.11` and `0.15.9`. We only bumped labels and versions in the `packages.xml` file with the new releases. Please see details in the git history https://github.com/ros2/rosbag2/commits/humble/mcap_vendor

closed 2024-08-10 by MichaelOrlov