ros2/rosbag2 #1692: [Documentation/Example] Please add example on how to read a rosbag with python

firesurfer, 2024-06-05:
## Description

There are currently only [examples](https://github.com/ros2/rosbag2/tree/rolling/rosbag2_examples/rosbag2_examples_py) on how to create a rosbag with python (and c++) but there are none on how to read a rosbag.  

## Related Issues

There is currently an open PR for adding a reader example in c++: https://github.com/ros2/rosbag2/pull/1683

## Completion Criteria

Example should show:

* How to open a rosbag
* Retrieve meta information 
* Read the data from the rosbag

Bonus:

* Convert some sensor data into an numpy array - I think this is the most common usecase when opening a rosbag in python

fujitatomoya, 2024-06-05:
I think this is reasonable and useful for beginners, in addition we can also add the documentation tutorial for this.

firesurfer, 2024-07-04:
In the mean time for whoever is looking for it: a probably less than perfect example on how to read a ros2bag with python:

```python
from rclpy.serialization import deserialize_message
import rosbag2_py
import sys
from pydoc import locate

def main():

    # Create a reader
    reader = rosbag2_py.SequentialReader()

    # Configure the storage we want to open
    storage_options = rosbag2_py._storage.StorageOptions(uri=sys.argv[1], storage_id="mcap")
    converter_options = rosbag2_py._storage.ConverterOptions("", "")

    # Actually open it
    reader.open(storage_options, converter_options)

    # Obtain a list of all topics and their types
    all_topics_and_types = reader.get_all_topics_and_types()

    print("Topics:")
    typestr_to_type = {}

    for meta_info in all_topics_and_types:
        print(" ", meta_info.name, meta_info.type)
        # Some python magic. We use the topic type to locate and import the python type
        typestr_to_type[meta_info.name] = locate(meta_info.type.replace("/", "."))

    # You can set a filter via:
    reader.set_filter(rosbag2_py._storage.StorageFilter(topics=["/joint_states"]))

    # Read it step for step
    while reader.has_next():
        next = reader.read_next()
        # Actually create the actual message
        msg = deserialize_message(next[1], typestr_to_type[next[0]])

        # NOTE: This is strictly sequential in the sense: as recorded.
        # This means read_next() will always return 1 (__one__) message!
        print(msg)

if __name__ == "__main__":
    main()

```

fujitatomoya, 2024-07-05:
related to https://github.com/ros2/ros2_documentation/issues/4560

fujitatomoya, 2024-08-01:
@clalancette can you close this too? https://github.com/ros2/ros2_documentation/issues/4560 closes this one too.

closed 2024-08-01 by clalancette