ros2/rosbag2 #1703: Move the `ReaderWriterFactory` to the `rosbag2_cpp` layer

MichaelOrlov, 2024-06-10:
## Description
Move the `ReaderWriterFactory` to the `rosbag2_cpp` layer

Currently the `ReaderWriterFactory` is part of the `rosbag2_transport` package and part of its namespace. 
However, the `ReaderWriterFactory` doesn't have any dependencies from the `rosbag2_transport` package.
It will be more appropriate if the `ReaderWriterFactory` will be located at the `rosbag2_cpp` package.

## Related Issues
- https://github.com/ros2/rosbag2/issues/1677
- https://github.com/ros2/ros2_documentation/pull/4499

## Completion Criteria

- [ ] The `ReaderWriterFactory` moved to the `rosbag2_cpp` layer and all dependencies in the rosbag2 updated.

## Implementation Notes / Suggestions

Since this is an API-breaking change, we can't just move `ReaderWriterFactory` from the `rosbag2_transport` namespace to the `rosbag2_cpp`. Instead, need to add a deprecation flag to the `rosbag2_transport::ReaderWriterFactory` and add a new `rosbag2_cpp::ReaderWriterFactory`.

## Testing Notes / Suggestions
Move relevant tests if they exist from the `rosbag2_transport` layer to the `rosbag2_cpp` layer.

Jeffxzj, 2025-01-23:
Kindly asked is there anyone working on this issue ? If not, i am glad to take it. @MichaelOrlov

MichaelOrlov, 2025-01-27:
@Jeffxzj Thank you for being willing to help with this task.  
No one has taken it so far. Feel free to take it.

Jeffxzj, 2025-02-07:
@MichaelOrlov  It seems that `ReaderWriterFactory` still depends on the `rosbag2_transport` package. The `rosbag2_transport::ReaderWriterFactory::make_writer` API needs the `rosbag2_transport::RecordOptions & record_options` as the function parameter. Since we need to keep the deprecated `rosbag2_transport::make_writer` API, the `rosbag2_transport` still need to depend on the `rosbag2_cpp` package. Adding `rosbag2_transport` to the `rosbag2_cpp` 's package dependencies can cause circular dependency error when `colcon build`

MichaelOrlov, 2025-02-09:
Hi @Jeffxzj,
The way how I envision a solution to break this cyclic dependency is the following
1. Will need to create a duplicate of the [CompressionOptions](https://github.com/ros2/rosbag2/blob/871a4477ee0dd714b2694e33fcc32abe436dbf15/rosbag2_compression/include/rosbag2_compression/compression_options.hpp#L59-L71) struct in the  `rosbag2_cpp` package and name it as `WriterOptions`. 
2. Use `rosbag2_cpp::WriterOptions` as the input parameter for the 
```cpp
   rosbag2_cpp::ReaderWriterFactory::make_writer(const rosbag2_cpp::WriterOptions & writer_options)
```
3. Copy values from the `writer_options` to the `rosbag2_compression::CompressionOptions compression_options` inside `ReaderWriterFactory::make_writer(..)`

MichaelOrlov, 2025-02-09:
- To avoid cyclic dependencies between `rosbag2_cpp` and `rosbag2_compression`, the `ReaderWriterFactory` should be somewhere outside of the `rosbag2_cpp` package.
- It seems that currently, due to the aforementioned cyclic dependencies, there is no better place for the `ReaderWriterFactory` rather than the `rosbag2_transport` package. Therefore, I am closing this task as it wouldn't be done.