ros2/rosbag2 #1727: Strange linking errors when building rosbag2_transport with GCC on Windows, Debug build

felixf4xu, 2024-06-26:
## Description
** clear and concise description of the bug is **

I'm building ROS2 on windows with GCC (actually mingw ucrt64, with gcc as the compiler) for a debug build, I have many link errors like this:

>[14/14] Linking CXX shared library librosbag2_transport.dll
FAILED: librosbag2_transport.dll librosbag2_transport.dll.a
C:\windows\system32\cmd.exe /C "cd . && C:\msys64\ucrt64\bin\c++.exe -g   -shared -o librosbag2_transport.dll -Wl,--out-implib,librosbag2_transport.dll.a -Wl,--major-image-version,0,--minor-image-version,0 @CMakeFiles\rosbag2_transport.rsp && cd ."
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/14.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/rosbag2_transport.dir/src/rosbag2_transport/player.cpp.obj:player.cpp:(.text+0x1627): undefined reference to `.refptr._ZNSt6chrono3_V212steady_clock3nowEv'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/14.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/rosbag2_transport.dir/src/rosbag2_transport/player.cpp.obj:player.cpp:(.text$_ZNSt9bad_allocC1Ev[_ZNSt9bad_allocC1Ev]+0x1b): undefined reference to `.refptr._ZTVSt9bad_alloc'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/14.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/rosbag2_transport.dir/src/rosbag2_transport/player.cpp.obj:player.cpp:(.text$_ZN10moodycamel17ReaderWriterQueueISt10shared_ptrIN15rosbag2_storage20SerializedBagMessageEELy512EEC1Ey[_ZN10moodycamel17ReaderWriterQueueISt10shared_ptrIN15rosbag2_storage20SerializedBagMessageEELy512EEC1Ey]+0x157): undefined reference to `.refptr._ZNSt9bad_allocD1Ev'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/14.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/rosbag2_transport.dir/src/rosbag2_transport/player.cpp.obj:player.cpp:(.text$_ZN10moodycamel17ReaderWriterQueueISt10shared_ptrIN15rosbag2_storage20SerializedBagMessageEELy512EEC1Ey[_ZN10moodycamel17ReaderWriterQueueISt10shared_ptrIN15rosbag2_storage20SerializedBagMessageEELy512EEC1Ey]+0x200): undefined reference to `.refptr._ZNSt9bad_allocD1Ev'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/14.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/rosbag2_transport.dir/src/rosbag2_transport/player.cpp.obj:player.cpp:(.rdata$_ZTIPDoFNSt6chrono10time_pointINS_3_V212steady_clockENS_8durationIxSt5ratioILx1ELx1000000000EEEEEEvE[_ZTIPDoFNSt6chrono10time_pointINS_3_V212steady_clockENS_8durationIxSt5ratioILx1ELx1000000000EEEEEEvE]+0x18): undefined reference to `typeinfo for std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long long, std::ratio<1ll, 1000000000ll> > > ()'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/14.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/rosbag2_transport.dir/src/rosbag2_transport/player.cpp.obj:player.cpp:(.rdata$_ZTISt5_BindIFMN6rclcpp16GenericPublisherEFvRKNS0_17SerializedMessageEESt10shared_ptrIS1_ESt12_PlaceholderILi1EEEE[_ZTISt5_BindIFMN6rclcpp16GenericPublisherEFvRKNS0_17SerializedMessageEESt10shared_ptrIS1_ESt12_PlaceholderILi1EEEE]+0x10): undefined reference to `typeinfo for std::_Weak_result_type<void (rclcpp::GenericPublisher::*)(rclcpp::SerializedMessage const&)>'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/14.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/rosbag2_transport.dir/src/rosbag2_transport/player.cpp.obj:player.cpp:(.rdata$_ZTIN6rclcpp7ServiceIN18rosbag2_interfaces3srv4StopEEE[_ZTIN6rclcpp7ServiceIN18rosbag2_interfaces3srv4StopEEE]+0x28): undefined reference to `typeinfo for std::enable_shared_from_this<rclcpp::Service<rosbag2_interfaces::srv::Stop> >'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/14.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/rosbag2_transport.dir/src/rosbag2_transport/player.cpp.obj:player.cpp:(.rdata$_ZTIN6rclcpp7ServiceIN18rosbag2_interfaces3srv4SeekEEE[_ZTIN6rclcpp7ServiceIN18rosbag2_interfaces3srv4SeekEEE]+0x28): undefined reference to `typeinfo for std::enable_shared_from_this<rclcpp::Service<rosbag2_interfaces::srv::Seek> >'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/14.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/rosbag2_transport.dir/src/rosbag2_transport/player.cpp.obj:player.cpp:(.rdata$_ZTIN6rclcpp7ServiceIN18rosbag2_interfaces3srv5BurstEEE[_ZTIN6rclcpp7ServiceIN18rosbag2_interfaces3srv5BurstEEE]+0x28): undefined reference to `typeinfo for std::enable_shared_from_this<rclcpp::Service<rosbag2_interfaces::srv::Burst> >'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/14.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/rosbag2_transport.dir/src/rosbag2_transport/player.cpp.obj:player.cpp:(.rdata$_ZTIN6rclcpp7ServiceIN18rosbag2_interfaces3srv8PlayNextEEE[_ZTIN6rclcpp7ServiceIN18rosbag2_interfaces3srv8PlayNextEEE]+0x28): undefined reference to `typeinfo for std::enable_shared_from_this<rclcpp::Service<rosbag2_interfaces::srv::PlayNext> >'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/14.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/rosbag2_transport.dir/src/rosbag2_transport/player.cpp.obj:player.cpp:(.rdata$_ZTIN6rclcpp7ServiceIN18rosbag2_interfaces3srv4PlayEEE[_ZTIN6rclcpp7ServiceIN18rosbag2_interfaces3srv4PlayEEE]+0x28): undefined reference to `typeinfo for std::enable_shared_from_this<rclcpp::Service<rosbag2_interfaces::srv::Play> >'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/14.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/rosbag2_transport.dir/src/rosbag2_transport/player.cpp.obj:player.cpp:(.rdata$_ZTIN6rclcpp7ServiceIN18rosbag2_interfaces3srv7SetRateEEE[_ZTIN6rclcpp7ServiceIN18rosbag2_interfaces3srv7SetRateEEE]+0x28): undefined reference to `typeinfo for std::enable_shared_from_this<rclcpp::Service<rosbag2_interfaces::srv::SetRate> >'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/14.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/rosbag2_transport.dir/src/rosbag2_transport/player.cpp.obj:player.cpp:(.rdata$_ZTIN6rclcpp7ServiceIN18rosbag2_interfaces3srv7GetRateEEE[_ZTIN6rclcpp7ServiceIN18rosbag2_interfaces3srv7GetRateEEE]+0x28): undefined reference to `typeinfo for std::enable_shared_from_this<rclcpp::Service<rosbag2_interfaces::srv::GetRate> >'

for ` undefined reference to '.refptr._ZNSt6chrono3_V212steady_clock3nowEv'`, it's more like a c++ compiler link error;

for ` undefined reference to 'typeinfo for std::_Weak_result_type<void (rclcpp::GenericPublisher::*)(rclcpp::SerializedMessage const&)>'`, not sure if it's related to `rclcpp::GenericPublisher`

and ` undefined reference to 'typeinfo for std::enable_shared_from_this<rclcpp::Service<rosbag2_interfaces::srv::Play> >'`.

### Actual Behavior
** clear and concise description of what _actually_ happened. include copied logs whenever possible **

I use the same compiler to build other ROS2 packages, more than 200 of them can be built on the same configuration (same `colcon build` command), but only `rosbag2_transport` has the link error.

If I switch the from `Debug` to `RelWithDebInfo`, it can build and link ok.

If I exclude `player.cpp` from the build files in cmake, it also builds ok without any linking error.

## System (please complete the following information)
 - OS: Windows
 - ROS 2 Distro: Jazzy

MichaelOrlov, 2024-06-26:
@felixf4xu If you see something like
> undefined reference to '.refptr._ZNSt6chrono3_V212steady_clock3nowEv'

It means that something more generic is wrong with your build setup.
At first glance, it looks like GCC is trying to mix and match libraries compiled with different mangling. It could be due to the mixing of debug libraries with non-debug. Or compile with different compiler versions or other flags.

Please double-check that you don't have leftovers in the `build` and `install` folders. Also, make sure that you **don't** source any binary releases before the `colcon build` command.
If you have `ccache` installed - I would recommend to clean its cache i.e., `ccache --clean`

felixf4xu, 2024-07-10:
it turns out to be related to the `-O0` settings, I'm building with `-DCMAKE_BUILD_TYPE=RelWithDebInfo` and ` -DCMAKE_CXX_FLAGS_RELWITHDEBINFO=-O0`.

Removing `-O0` fixed the issue. I would say it's mingw gcc related.

closed 2024-07-10 by felixf4xu