ros2/rosbag2 #1728: ros2 bag play error

qxanggggg, 2024-06-26:
I can't play my bag in my machine.Is anyone knows what would cause the problem?
unitree@ubuntu:~/Desktop/data_file$ ros2 bag play imu_data
[INFO] [1719402363.399387869] [rosbag2_storage]: Opened database 'imu_data/rosbag2_2024_06_15-18_50_19_0.db3' for READ_ONLY.
Segmentation fault (core dumped)
[image]

MichaelOrlov, 2024-06-26:
Please use our issue templates for reporting a bug or when creating a new issue.

There is not enough information to even start guessing what is going on.

MichaelOrlov, 2024-07-16:
- Closing as a stale issue without a response from the author.

closed 2024-07-16 by MichaelOrlov