ros2/rosbag2 #1730: ros2 bag play not working - "Opened database '<bagname>.db3' for READ_ONLY

juliaivonne, 2024-06-26:
## Description
I am trying to play a ROS2 bag folder which contains a metafile.yaml-file and a bag.db3-file. I am trying to play the bag in ROS2 (Ubuntu 20.04, ROS2 Foxy) with the command _ros2 bag play test3 (where _test 3__ is the name of the bag), but for some reasons the topics won't get published.

### Expected Behavior
The topics are published and I can use them, for example in rviz2.

### Actual Behavior
When I am trying to publish the topics with `ros2 bag play test3m where _test 3_`, I get the respond:

> [INFO] [1719414561.512699846] [rosbag2_storage]: Opened database 'test3/test3.db3' for READ_ONLY.

Also, when I am typing `ros2 topic echo <topicname>`, ROS2 says:

> WARNING: topic [<topicname>] does not appear to be published yet
> Could not determine the type for the passed topic.

The bagfile itself seems fine to me. The metafile.yaml-file makes sense and I get info from `ros2 bag info test3`:

Files:             test3.db3
Bag size:          64.2 MiB
Storage id:        sqlite3
Duration:          22.360s
Start:             Mar  1 2024 11:28:55.934 (1709288935.934)
End:               Mar  1 2024 11:29:18.294 (1709288958.294)
Messages:          3355
Topic information: Topic: <topicname> | Type: ros_can_gps_msg/msg/gpsData | Count: 2237 | Serialization Format: cdr
                   Topic: <topicname> | Type: sensor_msgs/msg/PointCloud2 | Count: 559 | Serialization Format: cdr
                   Topic: <topicname> | Type: visualization_msgs/msg/MarkerArray | Count: 559 | Serialization Format: cdr

Also, the topics are not shown in the topic list in rviz2.

I used 2 different ROS2-bags. One is a converted one from a ROS1-bag (with https://github.com/rpng/rosbags). The other one is a recorded turtlesim bag (I followed https://docs.ros.org/en/foxy/Tutorials/Beginner-CLI-Tools/Recording-And-Playing-Back-Data/Recording-And-Playing-Back-Data.html). I get the same results.

I installed Ubuntu & ROS again on another laptop, but nothing changed. 

## To Reproduce
Run  `ros2 bag play <any ROS2 bag>`

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: Foxy
 - Install Method: Debian (https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html), Deskop Version

## Additional context
** I am complete newbie to ROS & Ubuntu :) Thank you!**

MichaelOrlov, 2024-06-26:
@juliaivonne I am sorry, but `Foxy` has reached the end of its life, and we don't support it anymore. 
Could you please try to reproduce this issue with Humble, Iron, Jazzy or Rolling?
Also, please use the  `--log-level debug` CLI option for `ros2 bag play` for more detailed log output.

MichaelOrlov, 2024-07-16:
- Closing as a stale issue without a response from author.

closed 2024-07-16 by MichaelOrlov