ros2/rosbag2 #1734: No metadata.yaml is created after recording is stopped

oschan77, 2024-06-27:
## Description
Hi, I am trying to use ros2 bag record (in a Docker container) and stop recording using os.kill or os.killpg (instead of Ctrl + C). However, no metadata.yaml is created after recording is stopped is this way. I have tried signal.SIGINT, signal.SIGTERM. Both do not work for me. How to solve this issue? Thanks!

### Expected Behavior
metadata.yaml should be created after the recording is stopped.

### Actual Behavior
No metadata.yaml is created after recording is stopped.

## To Reproduce
1. from ScanRobotControls import ScanRobotControls
2. controls = ScanRobotControls()
3. controls.record()
4. controls.stop()

## System
 - OS: Ubuntu 22.04
 - ROS 2 Distro: Humble
 - Install Method: [APT, release archive, source]
 - Version: [release, branch, commit hash, patch release number]

## Additional context
```
import glob
import os
import signal
import subprocess
import time

class ScanRobotControls:
    def __init__(self):
        self.rosbag_dir_path = "/home/u/rosbag"
        self.proc_record = None

    def record(self):
        cmd_record = "docker exec rosbag bash -c 'source /opt/ros/humble/setup.bash; ros2 bag record /ouster/points /ouster/imu /ouster/scan /odom'"
        self.proc_record = subprocess.Popen(
            cmd_record,
            shell=True,
            stdin=subprocess.PIPE,
            stdout=subprocess.PIPE,
            stderr=subprocess.PIPE,
            preexec_fn=os.setsid,
        )

        self.proc_record = subprocess.Popen(
            cmd_record,
            shell=True,
            stdout=subprocess.PIPE,
        )

        print(f"Started rosbag record with pid {self.proc_record.pid}")

    def stop(self):
        if self.proc_record is not None:
            print(f"Stopping rosbag record with pid {self.proc_record.pid}")
            os.killpg(os.getpgid(self.proc_record.pid), signal.SIGINT)

            time.sleep(5)

            self.proc_record.wait()

            if self.proc_record.returncode is not None:
                print(
                    f"rosbag record process terminated with return code {self.proc_record.returncode}"
                )
            else:
                print("rosbag record process did not terminate properly.")
```

MichaelOrlov, 2024-06-27:
@oschan77 The behavior that you observe likely relates to how `SIGINT` is delivered to the docker container.
In your example, you are sending `SIGINT` to the docker container itself and not to the actual process with the rosbag2 recorder inside it.

We have integration tests to make sure that the rosbag2 recorder correctly handles the `SIGINT` signal.
Please see the https://github.com/ros2/rosbag2/blob/humble/rosbag2_tests/test/rosbag2_tests/test_rosbag2_record_end_to_end.cpp#L122-L181 for details.

This is not an rosbag2 issue.

fujitatomoya, 2024-06-27:
you are sending the SIGKILL to docker process not `ros2 bag record` process ID, which is generated in the another namespace created by dockerd.

i would handle the container hash id returned by docker command and call `docker stop xxx` instead. that is gonna send the SIGINT to the container and forward the signal to the executables running in the container. in that case, i am expecting that `ros2 bag recorder` process can receive the SIGINT and signal handler can manage that incoming signal.

see more details https://docs.docker.com/reference/cli/docker/container/run/#stop-timeout

I agree with @MichaelOrlov this is not the issue for rosbag2, i guess this issue can be closed.

closed 2024-06-28 by MichaelOrlov