ros2/rosbag2 #1736: Strange location of recorder and player executables

Timple, 2024-06-27:
## Description
We can run the recorder as a node by:
```
ros2 run rosbag2_transport recorder
```
This cannot be guessed but has to be reverse engineered from code.

### Expected Behavior
```
ros2 run rosbag2 recorder
```
or 
```
ros2 run ros2bag recorder
```

### Actual Behavior
Type `ros2 run rosbag2` and try tab-complete, this fails

## To Reproduce
These templates are getting rediculous.

## System (please complete the following information)
Jazzy

## Additional context
Yes, there the commanline cli of course. But in launchfiles this:

```XML
<node pkg="rosbag2" exec="recorder" />
```
Looks much better than
```XML
<executable cmd="ros2 bag record"/>
```
As it's consistent with other stuff, better passes ros arguments etc. Also node renaming and remapping is much better.

roncapat, 2024-06-27:
Hi @Timple, see https://github.com/ros2/rosbag2/pull/1510#discussion_r1470136830 for previous discussions on this matter.

roncapat, 2024-06-27:
And here is the draft PR https://github.com/ros2/rosbag2/pull/1603

Timple, 2024-06-27:
Great!

Do you want me to keep it open as placeholder? Or feel free to close it

roncapat, 2024-06-27:
I linked resolution of my PR to this issue, I would leave this open.
Could you check the PR and provide feedback, if any?
So I may try to find some time to complete it :)