ros2/rosbag2 #1757: Cannot Play ros2bag:  No storage id specified, and no plugin found that could open URI

Mysterium-sch, 2024-07-23:
I have this rosbag recorded and know that it is 25 G but when I try to run ros2 bag info, ros2 bag play or ros2 bag reindex, I get this error:
[ERROR] [1721742210.445309997] [rosbag2_storage]: No storage id specified, and no plugin found that could open URI

I get this error occasionally when creating a bag file in a launch file with this command:
```
    topics = [
        'jetson_1/image/compressed',
        'jetson_1/bar30/depth',
        'jetson_1/bar30/pressure',
        'jetson_1/bar30/temperature',
        'jetson_1/imagenex831l/range',
        'jetson_1/imu/data',
        'jetson_1/ekf/status',
        'jetson_1/imagenex831l/range_raw'
    ]

    return LaunchDescription(
        nodes + [
            ExecuteProcess(
                cmd=['ros2', 'bag', 'record', '-o', ct_str] + topics,
                output='screen'
            )
        ]
    )
```
Out of three runs on two devices, 2/6 of the bag files have this error.

Any insight on how to fix this would be greatly appreciated.

MichaelOrlov, 2024-07-24:
@Mysterium-sch Please provide additional information about what ROS 2 distro and OS version you use when observing this behavior?

Also, could you please try to reproduce this issue with `ros2 bag play` and the additional `--log-level debug` command line option and provide us with the captured log?

Mysterium-sch, 2024-07-24:
I am using Ubuntu 22 and Ros Humble

heres the output 

`[DEBUG] [1721823941.183356393] [rclcpp]: signal handler installed
[DEBUG] [1721823941.183449064] [pluginlib.ClassLoader]: Creating ClassLoader, base = rosbag2_storage::storage_interfaces::ReadWriteInterface, address = 0x63d27a6326d0
[DEBUG] [1721823941.183513238] [pluginlib.ClassLoader]: Entering determineAvailableClasses()...
[DEBUG] [1721823941.183529643] [pluginlib.ClassLoader]: Processing xml file /opt/ros/humble/share/rosbag2_storage/test/rosbag2_storage/test_plugin.xml...
[DEBUG] [1721823941.183512589] [rclcpp]: deferred_signal_handler(): waiting for SIGINT/SIGTERM or uninstall
[DEBUG] [1721823941.183579268] [pluginlib.ClassLoader]: XML file specifies lookup name (i.e. magic name) = my_test_plugin.
[DEBUG] [1721823941.183830211] [pluginlib.ClassLoader]: XML file specifies lookup name (i.e. magic name) = my_read_only_test_plugin.
[DEBUG] [1721823941.183840244] [pluginlib.ClassLoader]: Processing xml file /opt/ros/humble/share/rosbag2_storage_default_plugins/plugin_description.xml...
[DEBUG] [1721823941.183922540] [pluginlib.ClassLoader]: XML file specifies lookup name (i.e. magic name) = sqlite3.
[DEBUG] [1721823941.183931542] [pluginlib.ClassLoader]: Exiting determineAvailableClasses()...
[DEBUG] [1721823941.183936912] [pluginlib.ClassLoader]: Finished constructring ClassLoader, base = rosbag2_storage::storage_interfaces::ReadWriteInterface, address = 0x63d27a6326d0
[DEBUG] [1721823941.183944723] [pluginlib.ClassLoader]: Creating ClassLoader, base = rosbag2_storage::storage_interfaces::ReadOnlyInterface, address = 0x63d27a64f380
[DEBUG] [1721823941.184006103] [pluginlib.ClassLoader]: Entering determineAvailableClasses()...
[DEBUG] [1721823941.184012362] [pluginlib.ClassLoader]: Processing xml file /opt/ros/humble/share/rosbag2_storage/test/rosbag2_storage/test_plugin.xml...
[DEBUG] [1721823941.184076401] [pluginlib.ClassLoader]: XML file specifies lookup name (i.e. magic name) = my_test_plugin.
[DEBUG] [1721823941.184083247] [pluginlib.ClassLoader]: XML file specifies lookup name (i.e. magic name) = my_read_only_test_plugin.
[DEBUG] [1721823941.184090384] [pluginlib.ClassLoader]: Processing xml file /opt/ros/humble/share/rosbag2_storage_default_plugins/plugin_description.xml...
[DEBUG] [1721823941.184131986] [pluginlib.ClassLoader]: XML file specifies lookup name (i.e. magic name) = sqlite3.
[DEBUG] [1721823941.184138401] [pluginlib.ClassLoader]: Exiting determineAvailableClasses()...
[DEBUG] [1721823941.184143126] [pluginlib.ClassLoader]: Finished constructring ClassLoader, base = rosbag2_storage::storage_interfaces::ReadOnlyInterface, address = 0x63d27a64f380
[DEBUG] [1721823941.184171668] [pluginlib.ClassLoader]: Creating ClassLoader, base = rosbag2_cpp::converter_interfaces::SerializationFormatConverter, address = 0x63d27a712d70
[DEBUG] [1721823941.184229378] [pluginlib.ClassLoader]: Entering determineAvailableClasses()...
[DEBUG] [1721823941.184236034] [pluginlib.ClassLoader]: Processing xml file /opt/ros/humble/share/rosbag2_cpp/test/rosbag2_cpp/converter_test_plugin.xml...
[DEBUG] [1721823941.184298699] [pluginlib.ClassLoader]: XML file specifies lookup name (i.e. magic name) = a_converter.
[DEBUG] [1721823941.184311691] [pluginlib.ClassLoader]: XML file specifies lookup name (i.e. magic name) = s_converter.
[DEBUG] [1721823941.184317288] [pluginlib.ClassLoader]: Exiting determineAvailableClasses()...
[DEBUG] [1721823941.184322541] [pluginlib.ClassLoader]: Finished constructring ClassLoader, base = rosbag2_cpp::converter_interfaces::SerializationFormatConverter, address = 0x63d27a712d70
[DEBUG] [1721823941.184343285] [pluginlib.ClassLoader]: Creating ClassLoader, base = rosbag2_cpp::converter_interfaces::SerializationFormatSerializer, address = 0x63d27a5893b0
[DEBUG] [1721823941.184380588] [pluginlib.ClassLoader]: Entering determineAvailableClasses()...
[DEBUG] [1721823941.184386426] [pluginlib.ClassLoader]: Processing xml file /opt/ros/humble/share/rosbag2_cpp/test/rosbag2_cpp/converter_test_plugin.xml...
[DEBUG] [1721823941.184443463] [pluginlib.ClassLoader]: XML file specifies lookup name (i.e. magic name) = a_converter.
[DEBUG] [1721823941.184474100] [pluginlib.ClassLoader]: XML file specifies lookup name (i.e. magic name) = s_converter.
[DEBUG] [1721823941.184481714] [pluginlib.ClassLoader]: Exiting determineAvailableClasses()...
[DEBUG] [1721823941.184488928] [pluginlib.ClassLoader]: Finished constructring ClassLoader, base = rosbag2_cpp::converter_interfaces::SerializationFormatSerializer, address = 0x63d27a5893b0
[DEBUG] [1721823941.184498243] [pluginlib.ClassLoader]: Creating ClassLoader, base = rosbag2_cpp::converter_interfaces::SerializationFormatDeserializer, address = 0x63d27a3cf150
[DEBUG] [1721823941.184537475] [pluginlib.ClassLoader]: Entering determineAvailableClasses()...
[DEBUG] [1721823941.184542921] [pluginlib.ClassLoader]: Processing xml file /opt/ros/humble/share/rosbag2_cpp/test/rosbag2_cpp/converter_test_plugin.xml...
[DEBUG] [1721823941.184599474] [pluginlib.ClassLoader]: XML file specifies lookup name (i.e. magic name) = a_converter.
[DEBUG] [1721823941.184606204] [pluginlib.ClassLoader]: XML file specifies lookup name (i.e. magic name) = s_converter.
[DEBUG] [1721823941.184611557] [pluginlib.ClassLoader]: Exiting determineAvailableClasses()...
[... cut ...]
[DEBUG] [1721823941.285666343] [rcl]: Shutting down ROS client library, for context at address: 0x63d27a6af130
[DEBUG] [1721823941.285763912] [rclcpp]: SignalHandler::uninstall(): notifying deferred signal handler
[DEBUG] [1721823941.285887600] [rclcpp]: deferred_signal_handler(): woken up due to SIGINT/SIGTERM or uninstall
[DEBUG] [1721823941.285912567] [rclcpp]: deferred_signal_handler(): signal handling uninstalled
[DEBUG] [1721823941.286066127] [rclcpp]: signal handler uninstalled
No storage could be initialized from the inputs.
`

MichaelOrlov, 2024-07-24:
@Mysterium-sch From the log, the most important part is 
```
[DEBUG] [1721823941.198055546] [pluginlib.ClassLoader]: Instance of type rosbag2_storage_plugins::SqliteStorage created.
[DEBUG] [1721823941.198064184] [rosbag2_storage]: Trying storage implementation 'sqlite3'.
[DEBUG] [1721823941.198298598] [rosbag2_storage]: Failed to open with implementation 'sqlite3'. Continuing loop.
[ERROR] [1721823941.198309118] [rosbag2_storage]: No storage id specified, and no plugin found that could open URI
```

The `sqlite3` storage plugin was found and loaded correctly. However, the rosbag2 failed to open the specified file.

Please double-check the URI you are feeding for the Rosbag2 player and that the `DB3` file really exists and can be opened for read operation on the file system.
If the recorded file and path to it are correct, please share the recorded file with the minimum size so we can reproduce your problem.

Mysterium-sch, 2024-07-25:
Everything looks fine from my side, I recorded some more bag files and came across the issue on some more of the bag files. Here is one of the bag files where I am having the issue. 

https://drive.google.com/file/d/1DPRHBSlFH8Fb9tkGiYVSa2hs1tLW4-jx/view?usp=sharing

Mysterium-sch, 2024-08-21:
Still looking for help with this

MichaelOrlov, 2024-08-21:
Hi @Mysterium-sch,
I've just tried `ros2 bag info -s sqlite3 2000-02-23-20_49_50_0.db3` with your file and got a following output
```
[ERROR] [1724229262.959290506] [rosbag2_storage]: Could not open '2000-02-23-20_49_50_0.db3' with 'sqlite3'. Error: Failed to setup storage. Error: Error when processing SQL statement. SQLite error (11): database disk image is malformed
[ERROR] [1724229262.959306349] [rosbag2_storage]: Could not load/open plugin with storage id 'sqlite3'
```
The real error is `Error: Failed to setup storage. Error: Error when processing SQL statement. SQLite error (11): database disk image is malformed`. it does mean that your database file is corrupted. Perhaps we need to do a better job for error handling, since we don't see the same error message output without explicitly specifying the `-s sqlite3` CLI option.

However, you can google about this error message and the different ways how to restore your db3 file. I quickly found a link https://support.storj.io/hc/en-us/articles/360029309111-How-to-fix-a-database-disk-image-is-malformed

MichaelOrlov, 2024-09-20:
- Closing as a stale issue, without response from author. Feel free to reopen if you think it make sense to discuss more.

closed 2024-09-20 by MichaelOrlov

YKJamesMoriarty, 2024-10-25:
@Mysterium-sch HI, I encounter this problem too,I want to ask that have you have overcome this error?