ros2/rosbag2 #1760: Add CLI option to set compression threads priority for Rosbag2 recorder

MichaelOrlov, 2024-07-25:
## Description

We need to add a CLI parameter for the `ros2 bag record` parameter to be able to set up compression threads priority.

## Related Issues
- Follow up on https://github.com/ros2/rosbag2/pull/1457

## Completion Criteria

The CLI parameter for compression threads priority exists and is covered by unit tests.

## Implementation Notes / Suggestions

We already have the corresponding member variable `RecordOptions::compression_threads_priority` which matches to the 
`CompressionOptions::thread_priority` in the `ReaderWriterFactory::make_writer(..)` method.

This is going to be a relatively low-effort and straightforward task since changes are expected to be only at the `ros2bag` and maybe `rosbag2_py` packages. Perhaps without breaking API/ABI compatibility, and it will be possible to backport to the Jazzy branch.

## Testing Notes / Suggestions

We will need to add new or modify existing test in the `ros2bag/test/test_recorder_args_parser.py` to verify the correctness of the CLI parameter parser with the new CLI option.

Also, it will be great to add one integration test to the `ros2bag/test/test_record.py` to simply start the rosbag2 recorder with the newly added parameter and verify the debug output in the console. To facilitate debug output verification in the test will need to add some `ROSBAG2_COMPRESSION_LOG_INFO_STREAM(..)` or `ROSBAG2_COMPRESSION_LOG_DEBUG_STREAM(..)` output in the constructor of the `SequentialCompressionWriter::SequentialCompressionWriter(..)` and run `ros2 bag record` with corresponding `--loglevel` parameter from integration test. Note. This is supposed to be just a sanity check that the recorder can start with the settled-up CLI option. No need to publish messages and do a real recording with verification.

MichaelOrlov, 2024-07-25:
@r7vme Feel free to take this issue if you have time for it.

r7vme, 2024-07-26:
@MichaelOrlov sure, you can assign to me. I'll work on this issue next week.

PR #1768 (2024-08-09): Add cli option compression-threads-priority
This PR adds `--compression-threads-priority` CLI parameter for the `ros2 bag record` command to be able to set up compression threads priority.

- Follow up on https://github.com/ros2/rosbag2/pull/1457
- Closes #1760
PR #1778 (2024-08-10): [jazzy] Add cli option compression-threads-priority (backport #1768)
This PR adds `--compression-threads-priority` CLI parameter for the `ros2 bag record` command to be able to set up compression threads priority.

- Follow up on https://github.com/ros2/rosbag2/pull/1457
- Closes #1760 <hr>This is an automatic backport of pull request #1768 done by [Mergify](https://mergify.com).
closed 2024-08-09 by MichaelOrlov