ros2/rosbag2 #1762: Add progress bar and timestamp when playing back a bag

Kaju-Bubanja, 2024-07-26:
## Description

Currently there is no timestamp displayed when playing a bag and it doesn't show how far in you are.
This makes analysing specific issues at a certain point in time harder

## Completion Criteria

The current timestamp is shown when playing a bag, together with a completion percentage

## Implementation Notes / Suggestions

Seems straight forward, but I'm not proficient in writing CLI tools

## Testing Notes / Suggestions

Just play any bag

nicolaloi, 2024-10-11:
@MichaelOrlov If it's okay, I’d like to try to implement displaying the timestamp along with the completion percentage (which I believe is the 'current duration out of the total bag length'), similar to how ROS 1 prints these info during rosbag play.

MichaelOrlov, 2024-10-12:
@nicolaloi Thanks, it would be nice to have. 
A draft PR is welcome.

closed 2025-02-24 by MichaelOrlov