ros2/rosbag2 #1764: Decoding RecordOptions from YAML conflicts with chrono::milliseconds

JoLichtenfeld, 2024-07-26:
## Description
Currently, for parsing a yaml node to RecordOptions, the key "topic_polling_interval"  of type _std::chrono::milliseconds_ throws an error. Only the key "milliseconds" can be parsed to _std::chrono::milliseconds_.

### Expected Behavior
I try to parse a yaml node `node` to `rosbag2_transport::RecordOptions` using this approach:
`ROSBAG2_TRANSPORT_PUBLIC YAML::convert<rosbag2_transport::RecordOptions>::decode(node, record_options);`
The yaml node looks like this:

```
...
topics: []
topic_polling_interval: 100
topic_types: []
...
```

### Actual Behavior
[This line](https://github.com/ros2/rosbag2/blob/0fcc839d4363855fe4a5c8693b5e16dfd64b9b34/rosbag2_storage/include/rosbag2_storage/yaml.hpp#L100) throws the following error:

> terminate called after throwing an instance of 'YAML::BadSubscript'
> what():  yaml-cpp: error at line 29, column 25: operator[] call on a scalar (key: "milliseconds")

The method expects the key "milliseconds". If I change the yaml node to
```
...
topics: []
topic_polling_interval: 
  milliseconds: 100
topic_types: []
...
```
it works. But I don't see a reason why the additional key should be necessary.

## System
 - OS: Ubuntu 24.04
 - ROS 2 Distro: jazzy

MichaelOrlov, 2024-07-26:
@r7vme could you please make a preliminary analysis for this issue? It could be a side effect from recently merged fix for duration timepoint conversion.

r7vme, 2024-07-26:
@MichaelOrlov I think related code was added in this PR https://github.com/ros2/rosbag2/pull/1419 . Do you know why we need `node["milliseconds"]` instead of just using the `node`?

```cpp
  static Node encode(const std::chrono::milliseconds & duration)
  {
    Node node;
    node["milliseconds"] = duration.count();
    return node;
  }
```

can be 

```cpp
  static Node encode(const std::chrono::milliseconds & duration)
  {
    return Node(duration.count());
  }
```

MichaelOrlov, 2024-07-27:
Ahh, no. Please disregard my previous comment.
Actually it looks like a WAD. Works As Designed. Topics interval explicitly in milliseconds to be clear that it is in milliseconds. And I believe it was made on purpose this way to avoid confusion in what measurement it is.

JoLichtenfeld, 2024-07-30:
Alright, if it was intended this way then it's fine.

MichaelOrlov, 2024-08-10:
- Closing as WAD (Works As Designed)

closed 2024-08-10 by MichaelOrlov