ros2/rosbag2 #1770: [Feature Request] Add getters to the python binding of rclcpp::QoS

authaldo, 2024-08-06:
## Description

In ROS2 iron and lower the TopicMetadata structure offered the available QoS profiles as string, thus, allowing the user to check e.g. for certain settings. In my case I am mostly interested in identifying topics with 'transient local' durability as these have to be treated differently in the post processing of rosbags.

With Jazzy the TopicMetadata structure now returns the python binded equivalent of `std::vector<rclcpp::QoS>` (introduced in abdc408).  An overview of the python QoS structure can be found [here](https://github.com/ros2/rosbag2/blob/990475c7e252c38ee62ab45b9139b46e92fd3360/rosbag2_py/rosbag2_py/_storage.pyi#L51) while the pybind implementation is located [here](https://github.com/ros2/rosbag2/blob/990475c7e252c38ee62ab45b9139b46e92fd3360/rosbag2_py/src/rosbag2_py/_storage.cpp#L191).
As far as I can tell this binding only offers setters to create QoS structures with certain settings. Hence, the request to extend the binding with the corresponding getters in order to allow users a better interaction with the QoS structure.

Alternate ways I considered:
TopicMetadata offers a comparison method but after digging into the implementation it turns out that this comparison considers everything except the offered QoS profiles.

## Related Issues

No related issues (as far as I can tell).

## Completion Criteria

Addition of some getter functions for accessing the QoS settings. As a concrete example I would like to be able to check whether the durability setting of the offered QoS profile is set to 'transient local'. However, accessing the remaining settings is of course desired as well.

## Implementation Notes / Suggestions

The [rclpy equivalent of the QoS structure](https://github.com/ros2/rclpy/blob/464a357a7d9e39a12aed966fa3ccb44d50bb6816/rclpy/rclpy/qos.py#L69) may be helpful as reference.

## Testing Notes / Suggestions

clalancette, 2024-08-06:
https://github.com/ros2/rosbag2/pull/1648 may be related.

authaldo, 2024-08-07:
Thanks for bringing this up, a conversion to the python equivalent would indeed be enough to fix this issue.

ahcorde, 2024-08-22:
PR is merged, feel free to reopen the issue if the feature is not complete or you miss something

closed 2024-08-22 by ahcorde