ros2/rosbag2 #1797: Sort topics by name when executing `ros2 bag info`

boxanm, 2024-09-01:
## Description

Currently, when executing ros2 bag info, topics are not ordered alphabetically.
While this is manageable for small bagfiles, we encounter files with hundreds of topics.
Even though we invested time in grouping topics into meaningful namespaces, topics within the same namespace are not gathered together, making the output hard to analyze efficiently.

## Related Issues
It might also be convenient to allow users to sort the output by other criteria, such as topic type or message count.
However, simple alphabetical ordering could become the default option.

[This issue](https://github.com/ros2/rosbag2/issues/51) also seems relevant as moving the console output `ros2 bag info` to the  Python layer would necessarily influence any work on topics ordering.

## Completion Criteria

- `ros2 bag info` outputs topics sorted alphabetically by their names.

MichaelOrlov, 2024-09-03:
cc: @r7vme It looks like a good simple task.

r7vme, 2024-09-04:
@MichaelOrlov I'll take a look

tmillner21, 2024-09-04:
May I work on this issue?

r7vme, 2024-09-04:
@tmillner21 go ahead :+1:

PR #1804 (2024-10-17): Add "--sort" CLI option to the "ros2 bag info" command
This PR adds the "--sort" CLI option to the "ros2 bag info" command. With "--sort" CLI option user will be able to sort topics and services by name, topic type or number of recorded messages.

* generate sorted indices for topics from multiple different options (topic name, type, count or serialization format)
* defaults to sorting by topic name
* use sorted indices rather than using strict increasing indices for printing out the topics
* potential use of different sorting options through CLI by passing sort_method argument to format_topic [... cut]
PR #1838 (2024-12-02): [jazzy] Add "--sort" CLI option to the "ros2 bag info" command (backport #1804)
This PR adds the "--sort" CLI option to the "ros2 bag info" command. With "--sort" CLI option user will be able to sort topics and services by name, topic type or number of recorded messages.

* generate sorted indices for topics from multiple different options (topic name, type, count or serialization format)
* defaults to sorting by topic name
* use sorted indices rather than using strict increasing indices for printing out the topics
* potential use of different sorting options through CLI by passing sort_method [... cut]
closed 2024-10-17 by MichaelOrlov