ros2/rosbag2 #1798: Receiving rosout messages from rosbag2_recorder when running rosbag2_player

scn-nmsc, 2024-09-03:
## Description
ros2 bag play /testbag results in "rosbag2_recorder" actively publishing discovery log messages.

### Expected Behavior
When starting rosbag2_player then "rosbag2_recorder" should not appear to be subscribing to messages according to /rosout topic.

### Actual Behavior
When starting rosbag2_player (ros2 bag play /testbag) then "rosbag2_recorder" appears in the /rosout to be subscribing to topics (e.g. rosbag2_recorder: All requested topics are subscribed. Stopping discovery...)

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. run "ros2 bag play /testbag"
2. and subscribe to /rosout topic

## System (please complete the following information)
 - OS: Ubuntu 24.04
 - ROS 2 Distro: Jazzy
 - Install Method: APT
 - Version: 0.26.4

MichaelOrlov, 2024-09-03:
@scn-nmsc The Expected behaviour is unclear
>  When starting rosbag2_player then "rosbag2_recorder" should not appear to be subscribing to messages according to /rosout topic.

In the form how it is written is impossible to implement in the current design because `rosbag2_recorder` can't know that messages on the `/rosout` topic coming from the `rosbag2_player`

fujitatomoya, 2024-09-03:
I am not sure what the problem is with this issue report... can you rephrase the issue header?

Are you saying that `ros2 bag play /testbag` should not play `rosout` topic according to the `To Reproduce` section in the issue header? if that is so, why do you not specify or exclude the topic to play? the same can be don on `ros2 bag record` as well, so user can have the full control what needs to be recorded and played?

MichaelOrlov, 2025-05-07:
- Closing as a stale issue with no answer from the author. Feel free to reopen if you think that this issue still makes sense.

closed 2025-05-07 by MichaelOrlov