ros2/rosbag2 #1801: Deserialize messages in bag using the message definitions in the bag

jackiescanlon, 2024-09-10:
## Description

Right now with `rosbag2_py` it seems that its possible to:
1. Get the message definitions stored in an `.mcap` in string format ([source](https://github.com/ros2/rosbag2/blob/efd8d39843ce47500fde900178755b7550a4be73/rosbag2_cpp/src/rosbag2_cpp/readers/sequential_reader.cpp#L200-L205))
3. Get the messages stored in an `.mcap` by deserializing with the associated message definition ([example](https://mcap.dev/docs/python/ros2_example#reading-messages))

However, I cannot find a way to connect the two things, i.e., deserialize the messages via the definitions in the bag. There doesn't seem to be a (simple) way to turn the string message definitions into actual message defs that the deserializer can use. All the examples I can find for deserialization in python involve using the message definitions available in the local environment, not the ones in the bag.

Let me know if I am overlooking some part of the API that does do this currently.

## Related Issues

1. I found https://github.com/ros2/rosbag2/issues/782 which seems to have covered the problem of the message definitions not being stored in bag files at all in ROS 2, but using the stored definitions for deserialization does not appear to be covered by that ticket.
2. There's some comments on https://github.com/ros2/ros2/issues/1159#issuecomment-1644790582 requesting this also, but as far as I can tell no separate ticket was ever created for this.
4. Those comments reference https://github.com/ros2/ros2/issues/1374. I didn't see anything obvious in there that seems to cover the issue I'm describing but perhaps I overlooked it? Not sure.

## Completion Criteria

* A python API exists with which the user can deserialize bag messages using definitions stored in the bag, with no reliance on local environment definitions (i.e., no need to use `rosidl_runtime_py` to grab local defs)

## Implementation Notes / Suggestions

None

## Testing Notes / Suggestions

None

MichaelOrlov, 2024-09-17:
@jackiescanlon 
- I would recommend to create an appropriate issue in the https://github.com/ros2/ros2/issues/ similar to the https://github.com/ros2/ros2/issues/1374 

- There is nothing to do from the Rosbag2 side. We just don't have API from the underlying ROS 2 core packages to be able to deserialize messages using saved type description from the https://github.com/ros2/ros2/issues/1159

jackiescanlon, 2024-09-18:
Thanks @MichaelOrlov , I've made https://github.com/ros2/ros2/issues/1603 accordingly.