ros2/rosbag2 #1802: Quit a rosbag2_transport::Player instance in ROS Humble

MaxFleur, 2024-09-11:
I am writing a C++-based UI tool for various ROS2 related functionalities, which currently is based on ROS2 **humble**. One of these functionalities is that users should be able to play or stream a ROS2 bag file. So instead of using the command line tool, users should be able to play the bag file using a UI only.
To create a player instance and play the bag file, my code roughly looks like this:
```
void
PlayBagThread::run()
{
    rosbag2_storage::StorageOptions storageOptions;
    rosbag2_transport::PlayOptions playOptions;
    // bag dir, loop option and other parameters are set here

    auto player = std::make_shared<rosbag2_transport::Player>(storageOptions, playOptions);
    player->play();
}
```

Judging by the documentation, the `play` function opens an asynchronous thread handling the bag playing.
The problem now occurs if I try to close, stop or shutdown the player. The asynchronous thread cannot be accessed and thus, not be shut down. If I try to quit or terminate the thread running the function above, the application will crash because the asynchronous thread is still active.
For ROS2 **Jazzy**, a `stop` function has been added. In humble however, no such function is available. The command line script for playing a bag file also seems to rely on using a complete program termination to stop the player:
```
player = Player(args.log_level)
try:
    player.play(storage_options, play_options)
except KeyboardInterrupt:
    pass
```

I am well aware that the `rosbag2_transport::Player` instance is not meant to be used in an API-like style, as is the rclcpp-library. However, I would like to be able to play a bag file without using a command line tool. Is there any alternative to achieve this for ROS humble?

MichaelOrlov, 2024-09-12:
@MaxFleur Yeah, on Humble everything is more brutal. 
https://github.com/ros2/rosbag2/blob/9f766bf4ed08d834f32ae8304532698a14971196/rosbag2_py/src/rosbag2_py/_transport.cpp#L142-L172
We can't add `stop()` method because we will need to add synchronization primitives. We can't do this because we need to keep API/ABI compatibility for already released distros.
Sorry about that.
I would advice to move on Jazzy.

MaxFleur, 2024-09-12:
Understood. Seems like there is no real solution for Humble then... :-\
In this case, I'll try it in the future with Jazzy. This issue can be closed then.

closed 2024-09-12 by MaxFleur