ros2/rosbag2 #1826: How to read a mcap file recorded with OMGIDL encoding using rosbag2

karthiknit, 2024-10-02:
@MichaelOrlov I have a mcap file that contains data for foxglove::SceneUpdate message type with OMGIDL encoding. IDL file for this type is available here- https://github.com/foxglove/schemas/tree/main/schemas/omgidl/foxglove.
Is it possible to read the mcap file in such a scenario using rosbag2? Also, Is there anyway to convert omgidl encoding mcap file to ros2msg or ros2idl encoded mcap file.

MichaelOrlov, 2024-10-03:
@karthiknit I am not familiar with Foxglove OMGIDL encoding. How is it different from the `ros2idl`?
The `ros2idl` is also a subset of the OMGIDL.
In theory, you should be able to open the recorded mcap file with the `rosbag2:cpp` or `rosbag_py` API. 
However, to be able to playback such a file, you will need to have built and sourced typesupports generated from message definitions stored in the local file system.

cc: @defunctzombie

defunctzombie, 2024-10-03:
Have you tried to read the file? It is unclear from your question if you simply want to read the file with one of the APIs or want to `ros2 bag play` the file. My advice would be to try the specific workflow you want and see what works and what doesn't and come back with specific questions about those workflows.

> Also, Is there anyway to convert omgidl encoded mcap file to ros2msg or ros2idl encoded mcap file.

In theory yes since you can read the file using one of the many mcap language libraries and produce a new mcap file. What you would need to do specifically depends on what is in your original file and what expectations you have of "reading the file with rosbag2".

MichaelOrlov, 2025-05-01:
- Closing as a stale issue with no response from the author.

closed 2025-05-01 by MichaelOrlov