ros2/rosbag2 #1830: [humble] rosbag2_storage links to deprecated target

PizzocaroSolomon, 2024-10-11:
Hello everyone,

We have a project containing targets linking to rosbag2_storage and the standard yaml-cpp library.  
During compilation, we  get the following CMake warning:

[image]

We're certain that all our targets are linking to yaml-cpp using the target `yaml-cpp::yaml-cpp`.

Inside "/opt/ros/humble/share/rosbag2_storage/cmake/export_rosbag2_storageExport.cmake" we noticed the following lines:

`set_target_properties(rosbag2_storage::rosbag2_storage PROPERTIES
  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include/rosbag2_storage"
  INTERFACE_LINK_LIBRARIES "pluginlib::pluginlib;rcpputils::rcpputils;rcutils::rcutils;yaml-cpp"
)
`

and substituting "yaml-cpp" with "yaml-cpp::yaml-cpp", the warning disappears. 

We are convinced that this is a dirty solution to our problem. 
Do you have any suggestions on what we can do to remove the warning cleanly? 
Is it possible to add this "fix" to one of the next releases of ROS2 humble?

thank you very much!

clalancette, 2024-10-11:
> and substituting "yaml-cpp" with "yaml-cpp::yaml-cpp", the warning disappears.

The problem, as far as I remember, is that on Humble we have to use the `yaml-cpp` target; the `yaml-cpp::yaml-cpp` target did not exist there.

Are you using a custom compiled version of `yaml-cpp`?

PizzocaroSolomon, 2024-10-11:
That is correct, we do have a compiled version (latest) of `yaml-cpp` in our system.  
Is ROS2 providing its own version (I see a `yaml_cpp_vendor`)?  Will that version be updated? 

thank you very much

clalancette, 2024-10-11:
> Is ROS2 providing its own version (I see a `yaml_cpp_vendor`)?

Sort of.  `yaml_cpp_vendor` is used to provide `yaml-cpp` on those platforms that don't have it (like Windows).  On Ubuntu, we actually use the [system version of the package](https://packages.ubuntu.com/jammy/libyaml-cpp-dev).

> Will that version be updated?

Only if Ubuntu updates it in Ubuntu 22.04 (which is highly unlikely).

PizzocaroSolomon, 2024-10-11:
Ok understood. Is there a way we can "force" rosbag2_storage to keep pointing to the system version of yaml-cpp and not the one we are installing? I guess one solution would be to install our custom version in a different way...