ros2/rosbag2 #1834: GitHub CI: broken rolling->master mirroring

christophebedard, 2024-10-13:
There's a GitHub workflow that mirrors `rolling` to `master`: https://github.com/ros2/rosbag2/blob/rolling/.github/workflows/mirror-rolling-to-master.yaml. This was created when the default/Rolling branch was switched from `master` to `rolling`.

The workflow has been failing for a while: https://github.com/ros2/rosbag2/actions/runs/11312031065/job/31458961112. And `master` is now 177 commits behind `rolling`: https://github.com/ros2/rosbag2/tree/master. This is because #1247 was mistakenly merged into `master` instead of `rolling`, although it was subsequently reverted in #1248. However, since `master` can no longer be fast-forwarded to `rolling`, the mirroring action fails.

Options:

1. Force-push `rolling` to `master` to allow the mirroring action to work again
2. Get rid of the mirroring action
    - We could also get rid of the `master` branch, but we don't have to

fujitatomoya, 2024-10-13:
@christophebedard 

> Force-push rolling to master to allow the mirroring action to work again

probably if we take this way, the other core repos should align the policy...? IMO, i don't really know why we want to keep the `master` branch anyway? i think it is because some CI/CD or downstream users would rely on the branch `master`, i guess. but with given time, i think we can remove the `master` now...

christophebedard, 2024-10-13:
Yeah I agree, this should be a ROS 2 core-wide thing.

Having the `master` branch was useful for some time after the switch to allow people to transition, but I think it's probably been enough time now.

clalancette, 2024-10-13:
> Having the `master` branch was useful for some time after the switch to allow people to transition, but I think it's probably been enough time now.

So I agree that we can remove the workflow that attempts to backport things.  In most cases it no longer works, and even when it does it isn't providing any value.

However, for historical purposes we should keep the `master` branches around.

PR #1835 (2024-10-14): Remove rolling->master mirroring workflow
Resolves #1834

As discussed in the above issue, remove the mirroring workflow, but keep the `master` branch.
closed 2024-10-14 by MichaelOrlov