ros2/rosbag2 #1847: Add Support for Selective Playback of Topics with Mixed Serialization Formats in MCAP Files

jespersmith, 2024-10-31:
### Summary

When logging data with a custom logger, I store both ROS2 data and additional custom data in MCAP files. Currently, attempting to play back these files with `ros2 bag play` results in an error due to mixed serialization formats. It would be useful if we could selectively play back compatible topics without encountering this issue.

### Current Behavior

Attempting playback with `ros2 bag play` returns the error:

> "Topics with different RMW serialization formats have been found. All topics must have the same serialization format."

Even if specific topics are specified with the `--topics` flag, `ros2 bag play` checks all topics, rather than just those specified.

### Desired Behavior

1. **Ignore topics with unknown serialization formats**: `ros2 bag play` should play back topics with compatible formats (like `cdr`) and skip topics with unknown formats.
2. **Selective topic playback**: When specific topics are listed with the `--topics` flag, only these topics should be validated and played back, ignoring the others.

### Completion Criteria

- The solution should allow for playback of compatible topics without errors.
- Topics specified with `--topics` should be the only ones validated for serialization format.

### Suggested Implementation

1. **Option 1**: Update `mcap_storage.cpp` to generate metadata only for topics with "known" formats, ensuring playback compatibility.
   - **Note**: This approach may need updates if new formats are introduced in the future.
   
2. **Option 2**: Modify the behavior of the `--topics` flag in `ros2 bag play` to validate only the specified topics.

### Testing

1. Generate an MCAP file with a mix of valid `cdr` messages and messages in an unknown format.
2. Confirm that `ros2 bag play`:
   - Ignores unknown formats and plays compatible data.
   - Plays only specified topics when using the `--topics` flag, without checking non-specified topics.

fujitatomoya, 2024-10-31:
IMO this sounds reasonable and more reliable behavior.

> Ignore topics with unknown serialization formats: ros2 bag play should play back topics with compatible formats (like cdr) and skip topics with unknown formats.

probably it should not ignore, but it can print the warning that this topic data is unknown serialization formats or something like that.

MichaelOrlov, 2024-10-31:
I agree that it is nice to have feature/behavior. 
As regards the implementation proposal. It should be generic and not specific to some storage plugin.
Therefore only option 2 is valid in this case.
> 2. Option 2: Modify the behavior of the --topics flag in ros2 bag play to validate only the specified topics.

PRs are welcome.

jespersmith, 2024-11-04:
I made a first pass on Option 2) https://github.com/ros2/rosbag2/pull/1851

jespersmith, 2024-11-14:
Is the approach I'm taking in the PR acceptable? 

Is there any documentation on running test tests and linters locally?