ros2/rosbag2 #1859: ros2 bag record -a will coredump my program(c++)

userguide-galaxea, 2024-11-18:
My program use c++ API to create a node, this node have 10 publishers(all publish message at 1000hz). When I use ros2 bag record -a or ros2 topic hz /topic, my c++ program will coredump

(gdb) bt
#0  0x00007ffff6ecfffc in ?? () from /opt/ros/humble/lib/libfastrtps.so.2.6
#1  0x00007ffff6e5e12b in eprosima::fastrtps::rtps::StatefulWriter::matched_reader_add(eprosima::fastrtps::rtps::ReaderProxyData const&) () from /opt/ros/humble/lib/libfastrtps.so.2.6
#2  0x00007ffff70076fd in eprosima::fastrtps::rtps::EDP::pairing_remote_reader_with_local_writer_after_security(eprosima::fastrtps::rtps::GUID_t const&, eprosima::fastrtps::rtps::ReaderProxyData const&) () from /opt/ros/humble/lib/libfastrtps.so.2.6
#3  0x00007ffff6ea22c5 in eprosima::fastrtps::rtps::RTPSParticipantImpl::pairing_remote_reader_with_local_writer_after_security(eprosima::fastrtps::rtps::GUID_t const&, eprosima::fastrtps::rtps::ReaderProxyData const&) () from /opt/ros/humble/lib/libfastrtps.so.2.6
#4  0x00007ffff71150b8 in eprosima::fastrtps::rtps::security::SecurityManager::discovered_reader(eprosima::fastrtps::rtps::GUID_t const&, eprosima::fastrtps::rtps::GUID_t const&, eprosima::fastrtps::rtps::ReaderProxyData&, eprosima::fastrtps::rtps::security::EndpointSecurityAttributes const&, bool) () from /opt/ros/humble/lib/libfastrtps.so.2.6
#5  0x00007ffff70069e5 in ?? () from /opt/ros/humble/lib/libfastrtps.so.2.6
#6  0x00007ffff7006c24 in eprosima::fastrtps::rtps::EDP::pairing_reader_proxy_with_any_local_writer(eprosima::fastrtps::rtps::GUID_t const&, eprosima::fastrtps::rtps::ReaderProxyData*) () from /opt/ros/humble/lib/libfastrtps.so.2.6
#7  0x00007ffff700f395 in eprosima::fastrtps::rtps::EDPBaseSUBListener::add_reader_from_change(eprosima::fastrtps::rtps::RTPSReader*, eprosima::fastrtps::rtps::ReaderHistory*, eprosima::fastrtps::rtps::CacheChange_t*, eprosima::fastrtps::rtps::EDP*, bool) () from /opt/ros/humble/lib/libfastrtps.so.2.6
#8  0x00007ffff700fa4f in eprosima::fastrtps::rtps::EDPSimpleSUBListener::onNewCacheChangeAdded(eprosima::fastrtps::rtps::RTPSReader*, eprosima::fastrtps::rtps::CacheChange_t const*) () from /opt/ros/humble/lib/libfastrtps.so.2.6
#9  0x00007ffff6e7b644 in eprosima::fastrtps::rtps::StatefulReader::NotifyChanges(eprosima::fastrtps::rtps::WriterProxy*) () from /opt/ros/humble/lib/libfastrtps.so.2.6
#10 0x00007ffff6e7bdeb in eprosima::fastrtps::rtps::StatefulReader::change_received(eprosima::fastrtps::rtps::CacheChange_t*, eprosima::fastrtps::rtps::WriterProxy*, unsigned long) () from /opt/ros/humble/lib/libfastrtps.so.2.6
#11 0x00007ffff6e7c311 in eprosima::fastrtps::rtps::StatefulReader::processDataMsg(eprosima::fastrtps::rtps::CacheChange_t*) () from /opt/ros/humble/lib/libfastrtps.so.2.6
#12 0x00007ffff6e89990 in eprosima::fastrtps::rtps::MessageReceiver::process_data_message_without_security(eprosima::fastrtps::rtps::EntityId_t const&, eprosima::fastrtps::rtps::CacheChange_t&, bool) () from /opt/ros/humble/lib/libfastrtps.so.2.6
#13 0x00007ffff6e92e7d in eprosima::fastrtps::rtps::MessageReceiver::proc_Submsg_Data(eprosima::fastrtps::rtps::CDRMessage_t*, eprosima::fastrtps::rtps::SubmessageHeader_t*, bool) const () from /opt/ros/humble/lib/libfastrtps.so.2.6
#14 0x00007ffff6e94e95 in eprosima::fastrtps::rtps::MessageReceiver::processCDRMsg(eprosima::fastrtps::rtps::Locator_t const&, eprosima::fastrtps::rtps::Locator_t const&, eprosima::fastrtps::rtps::CDRMessage_t*) () from /opt/ros/humble/lib/libfastrtps.so.2.6
#15 0x00007ffff6e9b5af in eprosima::fastrtps::rtps::ReceiverResource::OnDataReceived(unsigned char const*, unsigned int, eprosima::fastrtps::rtps::Locator_t const&, eprosima::fastrtps::rtps::Locator_t const&) () from /opt/ros/humble/lib/libfastrtps.so.2.6
#16 0x00007ffff6f2b396 in eprosima::fastdds::rtps::UDPChannelResource::perform_listen_operation(eprosima::fastrtps::rtps::Locator_t) () from /opt/ros/humble/lib/libfastrtps.so.2.6
#17 0x00007ffff6f258db in ?? () from /opt/ros/humble/lib/libfastrtps.so.2.6
#18 0x00007ffff7adc253 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6

MichaelOrlov, 2024-11-23:
Hi @userguide-galaxea, from your description it is very unclear what setup do you have and how to reproduce the issue.

Please provide more information about your setup and how to reproduce this issue or provide a minimum valuable reproducer.

MichaelOrlov, 2025-03-01:
- Closing as a stale issue with no response from author

closed 2025-03-01 by MichaelOrlov