ros2/rosbag2 #1877: [jazzy] Trigger new release so fix Snapshot regression

doisyg, 2024-12-06:
## Description
Last released version in Jazzy is `0.26.5` and is from September; and it contains a bug that blocks our migration from Iron.
That bug makes the snapshot feature breaks after a couple of triggers BUT seems to be fixed with the latest Jazzy from source.
Hence, can we trigger a new Jazzy release please ? (and then I will ask for a Jazzy sync) @clalancette @MichaelOrlov 
cc @kaichie

clalancette, 2024-12-06:
We are planning to do a Jazzy sync in the next couple of weeks, it will get released then.  @marcoag FYI.

MichaelOrlov, 2024-12-18:
@marcoag Friendly ping here to not forget to make a release for Rosbag2 Jazzy and include it in the next synch/patch release.

marcoag, 2024-12-18:
Thanks for the reminder, working on it right now! 😉

MichaelOrlov, 2025-01-10:
- Closing as done.

closed 2025-01-10 by MichaelOrlov